tessel/t2-firmware #15: Node support for sysfs LEDs

kevinmehall, 2015-04-10:
The `require('tessels').led` API should use the LEDs attached to the MT7620n by writing to `/sys/class/leds/*/brightness`.

In addition to direct control, we could consider APIs for setting triggers.

johnnyman727, 2015-05-06:
@kevinmehall what did you have in mind for the triggering API?

kevinmehall, 2015-05-06:
```
tessel.led.blue.trigger('timer', { delay_on: 400, delay_off: 200 })
tessel.led.amber.trigger('netdev', { dev: 'wlan0' })
```
http://wiki.openwrt.org/doc/uci/system#leds
Since we set up the amber LED for wlan0 trigger by default, this may not be necessary. However, there are other triggers for USB activity, flash read/write, system load, etc.

johnnyman727, 2015-08-18:
Fixed with #52.

closed 2015-08-18 by johnnyman727