tessel/t2-firmware #21: Node support for PWM output

johnnyman727, 2015-04-29:


Frijol, 2015-06-20:
Are there more details associated with this issue? Would be good to get this clearly defined such that someone could work on it.

kevinmehall, 2015-06-20:
You'd need to look at the T2 schematics and SAMD21 datasheet for the timer compare output pins, add commands to port.c in the firmware, and then add APIs to use those commands in tessel.js / tessel.rs / etc.

kpihus, 2015-09-04:
As i understand, PWM support fot T2 is not ready yet ?
When it's scheduled to release ?

rwaldron, 2015-09-04:
@kpihus there is currently no schedule for PWM support

johnnyman727, 2015-09-04:
I'd love to get it in before we ship, but the list of to-dos is long :/

rwaldron, 2015-09-04:
It's pretty easy to connect a PCA9685 and get 16 PWMs for the price of two IO pins ;)

flaki, 2015-09-05:
We would still need a library wrapping the hardware - ideally I'd say we could support using the SAMD21 itself, or an external PWM controller (like the PCA9685 Rick mentioned above), but provide the APIs for setting it up properly.

I'm not all-that-well-versed in current solutions of the above (I know the [Filesystem API](https://bugzilla.mozilla.org/show_bug.cgi?id=1141021) proposed has PWM in mind but not sure if they have fleshed out any of the PWM support), so suggestions welcome. WiringPI is oft-cited in this relation, could we used some of those ideas?

Frijol, 2015-09-16:
The servo module is PCA9685.. does this fully cover using it as a PWM workaround? https://github.com/tessel/servo-pca9685#api-servo-setDutyCycle-whichServo-on-callback-Sets-the-duty-cycle-for-the-specified-servo-on-is-duty-cycle-uptime-range-from-0-1

johnnyman727, 2015-09-17:
It can work for now I suppose.

bruceme, 2015-10-14:
I have a project that require 1khz variable duty cycle.  I can bit-bang for now, but this library would really help. Any progress?

bruceme, 2015-10-14:
It appears there is at least some support for PWM now.

var tp = require('tessel-pwm');
var tessel = require('tessel');
var gpio = tessel.port['GPIO'];
var myPin = gpio.pwm[0];

gpio.pwmFrequency(1000);
myPin.pwmDutyCycle(0.75);

Source; https://github.com/eiriksm/tessel-pwm

This could be altered to support most simple PWM functions like; LED dimmers, directly driving servo signals, square wave generator.

rwaldron, 2015-10-14:
@bruceme that appears to be for the Tessel 1. 

> I have a project that require 1khz variable duty cycle. I can bit-bang for now, but this library would really help. Any progress?

The PCA9685 has an adjustable frequency range from ~24Hz to ~1.5kHz

tcr, 2015-11-03:
Dropping some links here:

* Timers reference: http://www.atmel.com/Images/Atmel-42181-SAM-D21_Datasheet.pdf#page=651
* SAMD21 firmware source: https://github.com/tessel/t2-firmware/tree/master/firmware

Frijol, 2016-01-21:
note courtesy of @mulderp: https://makerblog.info/2016/01/pwm-and-node-js-and-obviously-raspberry-pi/

johnnyman727, 2016-03-20:
I have looked further into PWM capabilities of the SAMD21. There are two peripherals that can modulate PWM signals easily: The TC (timer/counter) and the TCC (timer/counter for control applications). They have slightly different modes of operation but are generally quite similar. Some pins on the module ports have both a TC capability *and* a TCC capability while others have *only* a TC capability. For the sake of simplicity, I'll refer to both the TC and TCC peripherals as TCC.

### How PWM Works
Each TCC is constantly counting up with each processor clock tick until it hits some number, *TOP*, at which point it resets to zero. With each clock tick, the TCC checks to see if the current tick number (which is somewhere between 0 and `TOP`) is equal to another register that has been set prior to enabling the TCC called `CC` (compare/capture). If the counter value is equal to this register value, the TCC can optionally switch the state of some output pin. The state of the output pin is reset when the counter hits `TOP`. In this way, the `TOP` number acts as the period of the PWM and the `CC` number acts as the duty cycle. If `CC` is 1/2 of `TOP`, you will have a 50% duty cycle. There is also many more possible configurations, like counting down, that I didn't mention. The figure below helps convey how this works (although it's not great because it shows the `CC` value changing which we would not do for standard PWM generation):
[image]

### Pin Compatibility
Now we get into the complicated stuff. As I mentioned before, each pin has a TC or a TCC assigned to it (sometimes both). However, there are only three TCCs and eight TCs. Each TC/TCC has a either 2 or 4 `CC` registers which are used to set the duty cycle. We should not expose an API to set the duty cycle of two pins with the same `CC` register (on the same TCC) because they will overwrite each other's settings. Additionally, each pin has a specific `WO` (write output) register that is specifically assigned to it. In that sense, we need to pick pins with completely unique combinations of CC/WO pins for each TCC/TC. One thing that was confused to me is that I thought there would be a direct correlation between CC and WO pins (ie CC1 outputs on WO1) but that doesn't seem to be the case. I had to determine the correct WO number experimentally for each module port pin (that means I tried to output PWM 8*4 = 32 times and noted which WO numbers work for which CC numbers for a given TCC). I've made a table of pin assignments below:

#### Port A
| Module Port Pin Name | SAMD21 Pin Reference | TC/TCC Option 1 | TC/TCC Option 2 |
|----------------------|----------------------|-----------------|-----------------|
| SCL                  | PA13                 | TCC2/WO[1]/CC1  | TCC0/WO[7]/CC3  |
| SDA                  | PA12                 | TCC2/WO[0]/CC0  | TCC0/WO[6]/CC2  |
| SCK                  | PB23                 | TC7/WO[1]       |                 |
| MISO                 | PB22                 | TC7/WO[0]       |                 |
| MOSI                 | PB02                 | TC6/WO[0]       |                 |
| G1/TX                | PA14                 | TC3/WO[0]       | TCC0/WO[4]/CC0  |
| G2/RX                | PA15                 | TC3/WO[1]       | TCC0/WO[5]/CC1  |
| G3                   | PB08                 | TC4/WO[0]       |                 |

#### Port B
| Module Port Pin Name | SAMD21 Pin Reference | TC/TCC Option 1 | TC/TCC Option 2    |
|----------------------|----------------------|-----------------|--------------------|
| SCL                  | PA09                 | TCC0/WO[1]/CC1  | TCC1/WO[3]/CC1     |
| SDA                  | PA08                 | TCC0/WO[0]/CC0  | TCC1/WO[2]/CC0     |
| SCK                  | PA05                 | TCC0/WO[1]/CC1  |                    |
| MISO                 | PA07                 | TCC1/WO[1]/CC1  |                    |
| MOSI                 | PA04                 | TCC0/WO[0]/CC0  |                    |
| G1/TX                | PA10                 | TCC1/WO[0]/CC0  | TCC0/WO[2]/CC2     |
| G2/RX                | PA11                 | TCC1/WO[1]/CC1  | TCC0/WO[3]/CC3 |
| G3                   | PA02                 |                 |                    |

### Pins to use in JS API
**I think for our first pass at a JS PWM API, we should expose the capability to set PWM on G1 and G2 of each module port (so 4 pins in total)**. This has the benefit of being easy to remember for Tessel users and easy to implement because it's only a single TCC instance. We also don't have to deal with the TC. If we need more PWM pins later and somebody feels sufficiently motivated to add it, the firmware could be expanded upon to use both the TC and TCC pins. Additionally, it makes the JavaScript API simpler as I'll talk about in the next section.

### The PWM JavaScript API
There are several factors that influence the Tessel 2 PWM API:
* The API used by Tessel 1
* The capabilities of the hardware
* The amount of time we're willing to invest to get it done

The Tessel 1 API was relatively simple. It has two function calls: `pin.pwmDutyCycle` and `port.pwmFrequency`. This was originally roughly based off of the Arduino's PWM API. I think it's important to have the same two basic function calls.

In terms of hardware capabilities, we can change the duty cycle for each pin we've chosen but changing the frequency of any will change the frequency for all because they share the same TCC with the same `TOP [... cut ...]
In conclusion, based on the typical Tessel user experience and the capabilities of the hardware, I think the Tessel PWM API should be:
```
tessel.pin.pwmDutyCycle(dutyVal) // Some value between 0 and 1 
tessel.pwmFrequency(freq) // some value between 734 and 10000
```
API consumers will also have access to:
```
tessel.port.A.digital[0].pwmSupported //true
tessel.port.A.digital[1].pwmSupported //true
tessel.port.B.digital[0].pwmSupported //true
tessel.port.B.digital[1].pwmSupported //true
```

@rwaldron @reconbot what do you think?

mulderp, 2016-03-21:
Regarding the JS API: I once made a small PWM "simulator" here: http://embeddednodejs.com/pwm/ --> I used "cycleWidth" and "dutyCycle" as main parameters. Another API interesting for reference might be this: http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_pwm.html

rwaldron, 2016-03-21:
- `pwmFrequency`: no doubt, this is the best choice. If you had suggested `pwmPeriod`, I would've suggested something like `pwmFrequency` :+1: 
  - I agree that this should be `Tessel.prototype.pwmFrequency`, for the reason you illustrated. 
  - The frequency range you've stated: "734 and 10000", does that preclude the safe operation of a servo (50Hz) or motor (490Hz)? I'm able to control either of these with the SparkFun ATSAMD21G dev board, so I suspect that I'm misunderstanding something.
- `pwmDutyCycle`: is there any concern with resolution loss in scaling a fractional 0-1 value to the appropriate cycle value?

johnnyman727, 2016-03-24:
[Sorry this message is so delayed, the internet has been down. I originally tried to post it 36 hours ago.]

Thanks for the suggestions @mulderp. I think our API is pretty similar to the one linked (without convenience functions for different time scales) with the exception of `.disable`. `.disable` is essentially setting the duty cycle to 0 but perhaps some folks would like to disable the TCC altogether. @rwaldron do you think we need a `.disable` method?

> The frequency range you've stated: "734 and 10000", does that preclude the safe operation of a servo (50Hz) or motor (490Hz)? I'm able to control either of these with the SparkFun ATSAMD21G dev board, so I suspect that I'm misunderstanding something.

This is a good point - I had forgotten that servos operated at such a low frequency. The main reason why I chose 733 as the lowest frequency is because 48000000 (the clock speed of the chip) divided by anything lower than 734 (which would get us the period) will be greater than 0xFFFF. We cannot store a period larger than 0xFFFF because that register is 16-bits wide and 0xFFFF is the largest possible unsigned 16-bit number. However, the way to get around this is to use a prescalar on the general clock (or, more precisely, a prescalar other than 1). The maximum prescalar is 1024 so in theory we could get our frequency down to 734/1024 = ~0.72 Hz. **I suggest we document the range at 1Hz - 10kHz because it's easier to remember, @rwaldron thoughts?**. 

> pwmDutyCycle: is there any concern with resolution loss in scaling a fractional 0-1 value to the appropriate cycle value?

We never had any complaints with this API on the Tessel 1. If I understand your question correctly, the maximum resolution loss we'll see is 1/PERIOD which is generally pretty small.

johnnyman727, 2016-03-24:
Also, I'm not sure how this behaviour would be different from T1 but I think *we should disable PWM after setting the frequency*. Apart from the fact that the TCC *requires* the output to be disabled when changing configuration like clock prescalars, it can be dangerous for the hardware to have a new period set without changing the duty cycle. For example, if I set the frequency to 980Hz and the duty cycle to 50%, I could be setting an LED to slightly dim to prevent it from frying itself (I'm out of resistors in this scenario). Then for whatever reason, I set the frequency to 490Hz, now the duty cycle is going to be 100% and the resistor could be fried.

The suggested usage should always be to set the frequency first and then set the duty cycle. If you change the frequency, you need to re-set the duty cycle.

johnnyman727, 2016-03-24:
Two more notes:
T1 **requires** you to set the frequency before setting the duty cycle. I can't remember why we did that, probably some sort of hardware protection, but I think we should do that here as well. 

Second, I understand your point about resolution now. When the frequency is high, say 10k, there only 6 or so different duty cycles we can do (0xFFFF/10k = ~6). This is the source of the strange behavior I mentioned earlier. Perhaps we should set the cap to 5k because at least then you have more than 10 different duty cycles. @rwaldron what do you think a good cap would be?

rwaldron, 2016-03-24:
I'm just getting a chance to review this and the actual patch now, so my response here is mostly to let you know that I'm doing that. I'm going to hold off on responding to the above until after I've had a chance to thoroughly review.

rwaldron, 2016-03-24:
>  I suggest we document the range at 1Hz - 10kHz because it's easier to remember, @rwaldron thoughts?.

I agree. We can revisit if anyone has any issues or compelling use cases

PR #155 (2016-03-30): [WIP] Adds PWM capabilities and tests
Pending API comments on https://github.com/tessel/t2-firmware/issues/21.

Fixes #21.

Docs at https://github.com/tessel/t2-docs/pull/46.
closed 2016-03-30 by rwaldron