tessel/t2-firmware #26: rawWrite is undefined

johnnyman727, 2015-04-30:
I get the following error when running the `ambient.js` example. I haven't investigated yet:

```
/tmp/app/node_modules/ambient-attx4/node_modules/attiny-common/lib/index.js:26
  this.irq = hardware.digital[2].rawWrite(false);
                                 ^
TypeError: undefined is not a function
    at new Attiny (/tmp/app/node_modules/ambient-attx4/node_modules/attiny-common/lib/index.js:26:34)
    at new Ambient (/tmp/app/node_modules/ambient-attx4/index.js:41:17)
    at Object.use (/tmp/app/node_modules/ambient-attx4/index.js:383:10)
    at Object.<anonymous> (/tmp/app/node_modules/ambient-attx4/examples/ambient.js:13:26)
    at Module._compile (module.js:444:26)
    at Object.Module._extensions..js (module.js:462:10)
    at Module.load (module.js:339:32)
    at Function.Module._load (module.js:294:12)
    at Function.Module.runMain (module.js:485:10)
    at startup (node.js:112:16)
```

rwaldron, 2015-07-14:
Looks like this landed here: https://github.com/tessel/t2-firmware/commit/4ad83aa0c678656e2aac6b1bedb29831c0ca0e26

johnnyman727, 2015-07-14:
Indeed!

closed 2015-07-14 by johnnyman727

rwaldron, 2015-07-14:
:clap: