tessel/t2-firmware #30: USB Daemon crashes with process responses larger than the max pipe buffer size

johnnyman727, 2015-05-12:
I caught this error when I tried to retrieve details about what networks were around in a crowded space.

Essentially, what happens is that the epoll indicates the child process has data on stdout to be read. All 4096 bytes are read from stdout into the internal buffer which causes [the `space_available` var to become 0 and it removes the child process file descriptor from epoll](https://github.com/tessel/t2-firmware/blob/master/soc/usbexecd.c#L524-L528). 

Just after reading the data from the process into the internal buf, we check and see if we can [write from the internal buffer to the CLI](https://github.com/tessel/t2-firmware/blob/master/soc/usbexecd.c#L532-L537). We then proceed to write it out to the CLI. Once all the data has been written to the USB connection, [we check if there is credit left on the pipe buffer](https://github.com/tessel/t2-firmware/blob/master/soc/usbexecd.c#L404-L407). There isn't, so we attempt to remove the pipe buffer from epoll again, which causes the crash.

johnnyman727, 2015-05-12:
Documenting my thinking below:

We absolutely want to stop polling when there is no more space available in the pipe buffer (ie in the first call to `delete_epoll` so that the function to handle data is no longer called. It is acceptable to continue reading data into the internal pipe buffer even if the CLI has no credit because it will just be sent when credit opens up. 

I'll remove the second instance of `delete_epoll` after writing data to the CLI.

johnnyman727, 2015-05-13:
After that change, I'm encountering a new issue. What happens is after all of the data is written, the pipe buffer credit goes to 0, and the CLI eventually grants more credit. When this happens, the [file descriptor is re-added to the epoll](https://github.com/tessel/t2-firmware/blob/master/soc/usbexecd.c#L434-L437). Then the process of writing data from stdout to the internal buffer and to the CLI happens again but the pipe buffer is never removed from epoll because the internal pipe buffer isn't filled up all the way this time. Then, the same credit situation happens again where credit is at 0 and so it tries to `re-add` the fd to the pipe buffer which crashes it.

johnnyman727, 2015-05-13:
Essentially, the case of adding the pipe buffer to the epoll is incorrect. The pipe buffer is removed from the epoll when there is no more space in the epoll so it should be added when that condition is reversed. (ie `pb->bufcount == PIPE_BUF` and `ack_size > 0`).

johnnyman727, 2015-05-13:
Getting closer. Now the issue is that the pipe buffer file descriptor is not added back to the epoll after it is completely drained. I need to figure out how to check if it *was* full and is no longer full to add it back in.

johnnyman727, 2015-05-13:
Now there seems to be two more problems which exhibits itself as not all data is being returned the second time I run `tessel wifi -l` on the CLI:

1. Once a stream returns `eof`, it isn't necessarily removed from epoll so it continues calling the handler func over and over again.
2. Eventually the `SIGKILL` signal is emitted and it waits for the `stdout` stream to finish sending its data before actually closing the process. However, the `SIGKILL` signal tells the CLI not to send anymore acks. Therefore, no more credit is added to the stream and it goes into an infinite loop of handling itself.

If the CLI doesn't tell the Daemon to close a process once it receives a SIGKILL, when should it? How will the CLI know that there is data sitting in the internal pipe buffer?

kevinmehall, 2015-05-13:
Once stdout or stderr have read 0 bytes, I think they should be removed from epoll; otherwise they will continue to trigger events as you noted. This could set a flag in the pipebuf indicating that after all the data is drained, they should notify the EOF.

After receiving notification of process completion, the CLI should continue to acknowledge received data as usual if the CLI wants to receive the full response. The process exit status is independent of the state of the pipes on the daemon side. Once the process has exited and both stdout and stderr have signaled EOF, nothing else can happen, and it should clean up after itself by sending the CLOSE command.

johnnyman727, 2015-05-13:
We're nearly there. The root bug was fixed and now the last remaining issue actually resides in the CLI implementation. The CLI need to know exactly when to tell the remote process to close. If my understanding is correct, it should close the remote process when all four of the streams have sent their EOF. Does that sound right, @kevinmehall?

kevinmehall, 2015-05-14:
Or at least when the CLI has received notification of process exit and stdout and stderr EOF. The stdin and control streams' EOF is controlled by the CLI. If the process waits for stdin EOF, it will simply not exit until it receives it.

PR #31 (2015-05-14): [WIP] Fixes USBDaemon crash on large child process writes
Fixes #30.
closed 2015-05-14 by kevinmehall