tessel/t2-firmware #40: Fix usbpipe buffer size

kevinmehall, 2015-05-20:
https://github.com/tessel/t2-firmware/blob/master/firmware/usbpipe.c#L3 is suboptimal.

The bridge can handle 255 byte chunks. Unfortunately, USB is optimized when transfers are a multiple of the endpoint size (64), but the bridge cannot handle 256 byte chunks due to its use of a 1-byte length field.

johnnyman727, 2015-09-04:
@kevinmehall any tips on how to fix this? I'd love to try to speed up the usb comms.

kevinmehall, 2015-09-05:
As every piece of this puzzle was basically developed to the point where it worked enough to move on to the next fire, there's a lot of low-hanging fruit to optimize throughput. I'm going to describe these in terms of the PC -> Tessel direction as that's probably the biggest bottleneck, but most of them apply to the other direction as well:

### Optimize transfer batching

Log the sizes of the [outbound USB transfers](https://github.com/tessel/t2-cli/blob/ddd1fa79263e9994ccec400bc2791dead327b63d/lib/usb_connection.js#L69), and make sure that you're filling 64-byte packets most of the time, and ideally, batching together transfers of a multiple of that packet size like 4096 bytes. If there are a lot of packets that are only a few bytes due to headers or whatever, that propagates through the whole chain and wastes a lot of time that could be used to send useful data. Add buffering or cork/uncork support to fill the packets.

### Overlap USB and SPI transfers

Right now the firmware [toggles between](https://github.com/tessel/t2-firmware/blob/ebb1d82c40c6beae3a9de1b5287d46f546b515f6/firmware/usbpipe.c#L35) waiting for a USB packet to be received, and then waiting for it to be sent over the SPI bridge. While it's waiting for USB, SPI sits idle, and while it's waiting for SPI, USB sits idle.

Now that we have plenty of RAM on the MCU, you could make this a ring buffer, of say, 8 packets in each direction.

Be careful with short packets (less than the full 64 bytes). It needs to keep track of their lengths to avoid sending on the garbage in the buffer after the valid packet data. If you did the previous step right, short packets should be rare and happen only at the end of transfers, so you could simplify things by tracking only one short packet, and waiting for it to be flushed before accepting new packets from USB, so you know it's at the end of the ring buffer.

It might be structured like:

```.c
#define OUT_RING_SIZE 8
#define PACKET_SIZE 64
u8 out_ring_buf[OUT_RING_SIZE][PACKET_SIZE];
u8 out_ring_count = 0; // Number of packets in the ring buffer
u8 out_ring_write_pos = 0; // Packet index in which we're currently receiving a packet, or will once it's free
u8 out_ring_read_pos = 0; // Packet index from which we're currently sending a packet, or will once it's filled.
u8 out_ring_short_packet = 0; // If nonzero, the ring ends with a short packet of this size
bool out_usb_pending = false;
bool out_bridge_pending = false;

// Received from USB, send to bridge
void pipe_usb_out_completion() {
    u32 len = usb_ep_out_length(USB_EP_PIPE_OUT);
    if (len < PACKET_SIZE) out_ring_short_packet = len;
    out_ring_write_pos = (out_ring_write_pos + 1) % OUT_RING_SIZE;
    out_ring_count += 1;
    out_usb_pending = false;
    out_ring_step();
}

// Finished sending on bridge, start receive from USB
void pipe_bridge_in_completion() {
    out_ring_read_pos = (out_ring_read_pos + 1) % OUT_RING_SIZE;
    out_ring_count -= 1;
    out_bridge_pending = false;
    out_ring_step();
}

void out_ring_step() {
    if (!out_usb_pending && out_ring_count < OUT_RING_SIZE && out_ring_short_packet == 0) {
        usb_ep_start_out(USB_EP_PIPE_OUT, out_ring_buffer[out_ring_write_pointer], PACKET_SIZE);
        out_usb_pending = true;
    }

    if (!out_bridge_pending && out_ring_count > 0) {
        u8 len = PACKET_SIZE;
        if (out_ring_count == 1 && out_ring_short_packet != 0) {
             len = out_ring_short_packet;
             out_ring_short_packet = 0;
        }
        bridge_start_in(BRIDGE_USB, out_ring_buffer[out_ring_read_pointer], len);
        out_bridge_pending = true;
    }
}

```
(I started to write pseudocode, but basically implemented it)

As an optimization, if there are 3 (or 4, ending with a short packet) contiguous packets in the ring buffer, they could be sent as one SPI bridge transfer to address the issue mentioned in the original post on this issue. 

### Optimize spid and usbexecd

I would be surprised if there weren't terrible bottlenecks here if you did the above.

johnnyman727, 2015-09-05:
@kevinmehall thanks for the thorough reply! This gives me a good starting point.

johnnyman727, 2015-10-03:
I did a quick audit and it looks like when we are deploying, the most common transfer sizes are 4 (probably an ACK that the daemon received data) and 255 (the actual data being transferred) because that is the [max write length](https://github.com/tessel/t2-firmware/blob/master/soc/usbexecd.c#L59) as defined by the daemon.

johnnyman727, 2015-10-27:
@kevinmehall out of curiosity, do you think it could be sped up by increasing the priority of the usb daemon/spi daemon processes?

kevinmehall, 2015-10-27:
It would only help when something else is hogging the CPU, but it's worth a try.

johnnyman727, 2015-12-04:
getting started on the ring buffer here: https://github.com/tessel/t2-firmware/pull/130/files. @kevinmehall if you get a chance, could you look over the comments I wrote to ensure I understand it? There are a few comments I ended in a question mark to indicate that I don't understand why you added a particular line of code.

johnnyman727, 2015-12-16:
All these suggestions have been taken into account with #103.

closed 2015-12-16 by johnnyman727