tessel/t2-firmware #45: Bridge data corruption

kevinmehall, 2015-06-17:
Seems to happen with lots of small packets, and is commonly triggered with node versions that don't support `cork` on domain sockets (1.4 did support it, 2.0.2 does not).

Probably the SYNC handler takes too long and the firmware is not ready for the SPI transfer.

We need a tool to spam the ports with random echo commands, and a tool to watch the logic analyzer trace on a firmware patched to toggle a GPIO in the IRQ and watch for timing violations.

rwaldron, 2015-07-18:
@kevinmehall where does this stand? Is there a bug filed with io.js?

kevinmehall, 2015-07-18:
The original problem that made me suspect data corruption on split packets was actually a tessel.js issue and solved by #48. There are no known data corruption bugs in the bridge, but definitely could use more testing, so I am closing this issue in favor of #70.

`cork` is a performance optimization, but shouldn't be required for correctness. There was an io.js [pull request](https://github.com/nodejs/io.js/pull/910), which I closed after they merged something equivalent. Then it got reverted because it broke something on Windows. I haven't followed progress on it recently.

closed 2015-07-18 by kevinmehall