tessel/t2-firmware #59: I2C API design issues

rwaldron, 2015-07-06:
- The internal I2C class https://github.com/tessel/t2-firmware/blob/master/node/tessel.js#L424-L466 is designed to provide a 1:1 relationship between an instance and a specific device at a specific address.
- The I2C class is exposed via a Port instance method: https://github.com/tessel/t2-firmware/blob/master/node/tessel.js#L215-L220 
  - This method only creates a new I2C instance if the instance's faux-private `_i2c` property hasn't been initialized and assigned a value (which is expected to be an instance of I2C)
  - The address provided as an argument (`addr`) to this method is forwarded on to the I2C class constructor, which is stored as the value of a property of the instance that was created. 
  - The instance created is then assigned as the value of the port's `_i2c` property.
  - This only works correctly the first time, because any future invocations of `port.I2C(address)` will see that `this._i2c` has been assigned and will return that.
  - The consequence illustrated: 

```js
var device1 = port.I2C(0x00);
var device2 = port.I2C(0x01);

device1 === device2; // true
device2.addr; // 0x00
```

rwaldron, 2015-07-06:
Incidentally, this issue also causes incompatibility with t1-firmware: https://tessel.io/docs/hardwareAPI#i2c

rwaldron, 2015-07-06:
Fixing this has revealed a second, more problematic issue: Defining a `new`-able function as the property of the Port's prototype: 

```js
Port.prototype.I2C = function (addr, mode) {
  if (!this._i2c) {
    this._i2c = new I2C({addr: addr, mode:mode}, this);
  }
  return this._i2c;
};
```
And even the fix: 

```js
Tessel.Port.prototype.I2C = function(addr, mode) {
  return new Tessel.I2C({
    addr: address,
    mode: mode,
    port: this
  });
};
```  

Are both wrong, because when a user writes: 

```js
new tessel.port.A.I2C(...);
```

The value of `this` within the scope of `tessel.port.A.I2C` (which is `Port.prototype.I2C`) is actually an instance of `Port.prototype.I2C`. Take a look at this reduced case: 

```js
function Port(id) {
  this.id = id;
}

function Protocol(port) {
  this.port = port;
}

Port.prototype.Protocol = function() {
  console.log(this instanceof Port.prototype.Protocol); 
  // true
  // 
  // This means that properties and characteristics of
  // the port instance that are expected to be passed on 
  // to Protocol are lost, because `this` is the wrong
  // object

  return new Protocol(this);
};

var port = new Port("A");
var protocol = new port.Protocol();

console.log(protocol.port.id); // undefined
```

rwaldron, 2015-07-06:
This can be fixed by using the same implementation approach that was used in t1-firmware/builtin/tessel.js: these instance-sensitive constructors are defined as own-instance properties, within the Port class constructor itself (since there are only two Port objects created, this change has negligible cost)

rwaldron, 2015-07-06:
https://github.com/rwaldron/t2-firmware/commit/bd3b3118af75f9ad30afd14160784929d2f138a0

rwaldron, 2015-07-06:
FWIW: I've been running this tessel.js in place of the tessel.js that shipped with the latest build.

johnnyman727, 2015-07-07:
Glad you caught this! All PRs have been reviewed and merged/commented.

closed 2015-07-08 by rwaldron