tessel/t2-firmware #71: Need better documentation on limitations for pins

Frijol, 2015-07-19:
E.g. only pins 2,5,6,7 on both ports support interrupts.

Other important information that needs documentation?

johnnyman727, 2015-09-01:
This could also be integrated into the firmware. ie if someone types `tessel.port.A.digital[0].once('high'...)` we should throw an error to notify them that interrupts aren't available on that pin.

rwaldron, 2015-09-01:
> we should throw an error to notify them that interrupts aren't available on that pin.

That's what happens already ;)

`tessel.port.A.digital[0]` won't trigger that error, because `tessel.port.A.digital[0]` points to `tessel.port.A.pin[5]` which is a valid interrupt pin. 

The following illustrates: 

```js
var tessel = require('tessel');

tessel.port.A.pin[0].once('high', function() {
  console.log('interrupted going high');
});
```

```
$ t2 run index.js
INFO Connecting to Tessel...
INFO Connected to bishop over LAN
INFO Writing index.js to RAM on bishop (3.072 kB)...
INFO Deployed.
INFO Running index.js...
/usr/lib/node/tessel-export.js:362
      throw new Error('Interrupts are not supported on pin ' + this.pin);
            ^
Error: Interrupts are not supported on pin 0
    at Tessel.Pin.addListener (/usr/lib/node/tessel-export.js:362:13)
    at once (events.js:265:8)
    at Object.<anonymous> (/tmp/remote-script/index.js:4:22)
    at Module._compile (module.js:426:26)
    at Object.Module._extensions..js (module.js:444:10)
    at Module.load (module.js:351:32)
    at Function.Module._load (module.js:306:12)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:117:18)
    at node.js:948:3
INFO Stopping script...
```

johnnyman727, 2015-09-04:
Ah my mistake! I mixed up `.pin` and .`digital`. Thanks @rwaldron!

Frijol, 2015-10-26:
ha, that's actually the problem, people have to guess and check to find that out.
Maybe we just need a better error message, like "that pin doesn't work but these other ones do"

Frijol, 2016-04-15:
this documentation error has been fixed a while ago: https://tessel.io/docs/hardwareAPI#ports-and-pins

Frijol, 2016-04-15:
reopening because we should give people a path forward in the error message

closed 2016-04-15 by Frijol