tessel/t2-firmware #72: Reading I2C creates RangeError: Maximum call stack size exceeded

Frijol, 2015-07-19:
Tried this with the color sensor & reported as a bug there: https://github.com/johnnyman727/rgb-tcs34725/issues/2

But hitting the same issue with the Gesture Sensor: https://github.com/jiahuang/apds-gesture

Both hit the issue while trying to receive data over I2C.

```sh

:0

RangeError: Maximum call stack size exceeded
```

rwaldron, 2015-07-20:
I have one of these sensors, so I will try to reproduce and report back

rwaldron, 2015-07-20:
I doubt this is related to your issue, but worth noting that this wiring isn't possible on T2:

```
VIN -> Unconnected
GND -> GND
3V3 -> 3V3
SCL -> SCL
SDA -> SDA
INT -> GPIO 1
LED -> GPIO 2
```

The last two need to be updated to:

```
INT -> GPIO 2
LED -> GPIO 3
```

rwaldron, 2015-07-20:
Running t2-firmware compiled from master, with the following program:

```js
var tessel = require('tessel');
var rgbLib = require('rgb-tcs34725');
var rgb = rgbLib.use(tessel.port.A);

rgb.on('ready', function() {
  console.log("rgb is ready");

  setInterval(function() {
    rgb.calculateLux(function(err, lux) {
      if (err) throw err;

      console.log('LUX:', lux);
    })
  }, 1000);
})
```

I get the this result: 

```
rwaldron at nova in ~/codez/tessel-issues/tcs34725
$ t2 run index.js
INFO Connecting to Tessel...
INFO Connected to arnold over LAN
INFO Writing index.js to RAM on arnold (123.904 kB)...
INFO Deployed.
INFO Running index.js...
rgb is ready
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1575.104
LUX: 1286.79936
LUX: 1575.104
LUX: 1625.571840000001
LUX: 2941.696
LUX: 1658.21696
LUX: 2347.264000000001
LUX: 29907.88608
LUX: 21039.88736
LUX: 15702.737920000005
LUX: 12946.229760000002
LUX: 9872.09216
^CINFO Stopping script...
```

Frijol, 2015-07-22:
Ok, will try again with updated firmware; maybe this is already fixed.

kevinmehall, 2015-07-22:
This looks like the issue fixed by #48

LinusU, 2015-09-02:
I think that this is fixed, the accelerometer module tests pass for me on my tessel 2.

rwaldron, 2015-09-02:
Confirmed!

closed 2015-09-02 by rwaldron