tessel/t2-firmware #74: SPI with default chip select

LinusU, 2015-07-23:
Trying to instantiate an SPI connection without specifying chipSelect fails with this error.

```text
usr/lib/node/tessel.js:464
    this.chipSelect = params.chipSelect || this._port.digital[0];
                                                             ^
TypeError: Cannot read property '0' of undefined
    at new SPI (/usr/lib/node/tessel.js:464:62)
    at new Port.SPI (/usr/lib/node/tessel.js:217:51)
    at new ISP (/tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:17:14)
    at Object.use (/tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:563:10)
    at Object.updateFirmware (/tmp/remote-script/node_modules/attiny-common/lib/attiny-isp.js:138:20)
    at Attiny.updateFirmware (/tmp/remote-script/node_modules/attiny-common/lib/index.js:181:7)
    at /tmp/remote-script/node_modules/attiny-common/lib/index.js:62:14
    at /tmp/remote-script/node_modules/attiny-common/lib/index.js:138:9
    at Port.spiComplete (/tmp/remote-script/node_modules/attiny-common/lib/index.js:173:19)
    at Port.<anonymous> (/usr/lib/node/tessel.js:93:40)
```

rwaldron, 2015-07-23:
Can you update your firmware to master and try this again?

Here's a script I use to update the on-board tessel.js module: 

```sh
#!/bin/sh

# Replace with your T2's name
NAME="arnold"
# Replace with your path to t2-firmware repo
LOCAL="/Users/rwaldron/clonez/t2-firmware"
# Don't change this
REMOTE="/usr/lib/node"

scp -i ~/.tessel/id_rsa $LOCAL/node/tessel.js root@$NAME.local:$REMOTE/tessel.js
```

kevinmehall, 2015-07-23:
Terminology note: "firmware" should probably refer to the code in flash on the SAMD21. `tessel.js` is in this `t2-firmware` repo for now because it's closely coupled to that firmware and needs to be kept in sync for protocol changes, but it's just a normal JS library. If you want to update the firmware on the SAMD21, you need to use its USB bootloader.

LinusU, 2015-07-24:
@rwaldron I just built master of this repo and flashed it to my tessel 2, shouldn't that update the tessel.js file?

Well it didn't seem like it so I manually copied it over, per your script, and it resolved the error.

closed 2015-07-24 by LinusU

kevinmehall, 2015-07-24:
Updating the firmware on the SAMD21 won't change a file on the Linux filesystem. `tessel.js` gets included in the OpenWrt image through the `tessel-tools` package, which [references](https://github.com/tessel/openwrt-tessel/blob/master/package/tessel/tools/Makefile#L10) a specific version of the `t2-firmware` repo.

For testing, it also works to `scp` the file to your Tessel.

rwaldron, 2015-07-24:
> Terminology note: "firmware" should probably refer to the code in flash on the SAMD21

Got it, agreed and thanks for establishing the concrete definition :)