tessel/t2-firmware #75: I2C TypeError on construction

LinusU, 2015-07-23:
I get this error when running `new hardware.I2C(0x24)`

```text
/usr/lib/node/tessel.js:430
    this._port._simple_cmd([CMD.ENABLE_I2C, this._baud]);
               ^
TypeError: this._port._simple_cmd is not a function
    at new I2C (/usr/lib/node/tessel.js:430:16)
    at new Port.I2C (/usr/lib/node/tessel.js:210:21)
    at new RFID (/tmp/remote-script/index.js:63:14)
    at Object.use (/tmp/remote-script/index.js:723:10)
    at /tmp/remote-script/test/readcard.js:15:13
    at /tmp/remote-script/node_modules/tinytap/src/tester.js:101:5
    at doNTCallback0 (node.js:408:9)
    at process._tickCallback (node.js:337:13)
    at Function.Module.runMain (module.js:469:11)
    at startup (node.js:117:18)
```

LinusU, 2015-07-23:
Hmm, seems like calling it without `new` works... This works in another module `hardware.I2C(0x1D)`

rwaldron, 2015-07-23:
What is `hardware`? Can you paste the rest of the code? Thanks!

LinusU, 2015-07-24:
Sorry, `hardware` is a tessel port.

```javascript
var tessel = require('tessel')

var hardware = tessel.port.A
var i2c = new hardware.I2C(0x24)
```

This was resolved by upgrading to the latest `tessel.js` from `master` thought.

closed 2015-07-24 by LinusU

rwaldron, 2015-07-24:
Ah, yes—I hit this same issue and patched it :)