tessel/t2-firmware #76: I2C send never calls callback

LinusU, 2015-07-23:
This is causing problems with the accel module. The callback is never called and `no` is never printed.

```javascript
console.log('yes')

i2c.send(new Buffer([0x2A, 0x00]), function (err) {
  console.log('no')
})
```

johnnyman727, 2015-07-26:
@kevinmehall any idea why this might be?

LinusU, 2015-09-02:
This is still a problem and blocks the RFID module (and probably lots of others)

johnnyman727, 2015-09-04:
This will be next on my todo. It's something we should have working by now.

johnnyman727, 2015-09-07:
@LinusU I can't reproduce this issue anymore. I'm not sure what might have fixed it though.

I have confirmed that the accelerometer works with T2.

johnnyman727, 2015-09-07:
Actually, before we close this. @LinusU can you confirm?

LinusU, 2015-09-07:
I could have sworn that it didn't work on the RFID module but that it did work for the Acceleration last time I checked (just a few days ago). But apparently it must have been another issue with the RFID since it logs the return of it's first `i2c.send` command.

Closing as fixed!

closed 2015-09-07 by LinusU

johnnyman727, 2015-09-07:
:clap: