tessel/t2-firmware #78: TypeError: immediate._onImmediate is not a function

LinusU, 2015-07-24:
Hitting this error when trying to use the ambient module.

```text
timers.js:371
      immediate._onImmediate();
                ^
TypeError: immediate._onImmediate is not a function
    at processImmediate [as _immediateCallback] (timers.js:371:17)
```

LinusU, 2015-07-24:
This seems to happen when you run `setImmediate` without a callback. That isn't happening in the ambient code thought.

LinusU, 2015-09-02:
This is still a problem with the latest version.

johnnyman727, 2015-09-04:
@LinusU weirdly, I get a slightly different error message when I run simply `setImmediate()` in a script:
```
timers.js:367
      immediate._onImmediate();
                ^
TypeError: undefined is not a function
    at processImmediate [as _immediateCallback] (timers.js:367:17
```

I wonder where `immediate._onImmediate` is defined in your scenario.

flaki, 2015-09-05:
`immediate._onImmedite()` is used by node/io.js to queue and execute [microtasks](). Apparently (at least on my board) we have `io.js` version `2.3.0`, but the latest version seems to be [`2.3.4`]() (that is, in the 2.3 branch).
The exception for me is thrown in [`timers.js:371`](https://github.com/nodejs/node/blob/v2.3.1-release/lib/timers.js#L371), but that doesn't seem overly helpful.

Could this be simply an outdated node version issue?

On the long term we should also update to (the unified) node.js 4.0 just recently (being) released, I think. Since that's going to be quite a big leap forward, that should happen the earliest as possible, to allow enough time for features to bake, bug to be fixed, since going forward I think 4.0 will be the de-facto baseline the T2 should be aiming for (especially since the pushed-back release date).

johnnyman727, 2015-09-05:
@flaki isn't Node on v0.12.7? 

I made an issue on the openwrt repo to update the running version of Node: https://github.com/tessel/openwrt-tessel/issues/22

johnnyman727, 2015-09-05:
Ah I see what you mean https://github.com/nodejs/node/releases

johnnyman727, 2015-10-18:
I confirmed that this is not resolved with an update to Node V4.1.2 :(

johnnyman727, 2015-10-18:
Fixed by https://github.com/tessel/avr-isp/pull/8. Not that ambient/ir still don't work due to https://github.com/tessel/t2-firmware/issues/112

closed 2015-10-18 by johnnyman727

LinusU, 2015-10-19:
Cool :