tessel/t2-firmware #79: TypeError: i2c._initialize is not a function

LinusU, 2015-07-24:
Seems like this function on the `I2C` class is no longer present.

```text
/tmp/remote-script/index.js:6
i2c._initialize()
    ^
TypeError: i2c._initialize is not a function
    at Object.<anonymous> (/tmp/remote-script/index.js:6:5)
    at Module._compile (module.js:426:26)
    at Object.Module._extensions..js (module.js:444:10)
    at Module.load (module.js:351:32)
    at Function.Module._load (module.js:306:12)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:117:18)
    at node.js:948:3
```

Testcase:

```javascript
var tessel = require('tessel')

var hardware = tessel.port.A
var i2c = new hardware.I2C(0x24)

i2c._initialize()
```

rwaldron, 2015-07-24:
Which module uses this? It should be updated to _not_ do this if `tessel.port.version` is 2, because this: https://github.com/tessel/t1-firmware/blob/master/builtin/tessel.js#L651-L666 happens in the I2C [constructor](https://github.com/tessel/t2-firmware/blob/master/node/tessel.js#L450-L467) (where it belongs).

johnnyman727, 2015-07-26:
Yes, we should just update the module driver to not use that function anymore.

LinusU, 2015-07-27:
This was from the RFID module, would this be good way of solving it?

```javascript
if (typeof i2c._initialize === 'function') {
  i2c._initialize()
}
``

johnnyman727, 2015-07-29:
Yeah, that looks good to me.

LinusU, 2015-07-29:
I submitted tessel/rfid-pn532#23 but realized that it might be better to fix it here as well by providing a no-op function with a deprecation warning? That way other modules that are trying to use this function should get a pointer on why it isn't working.

johnnyman727, 2015-07-30:
At some point we just need to pull deprecated code and the switch from T1 to T2 is probably as good of a time as any.

johnnyman727, 2015-09-07:
Closing as a will not fix.

closed 2015-09-07 by johnnyman727