tessel/t2-firmware #82: Expose enabling/disabling individual module ports on the T2

flaki, 2015-08-03:
I noticed that the module ports only fire up once the `require("tessel")` call instantiates the hardware API. After this happens the module ports power up, and LEDs on both sides of the POWER led (next to their respective module ports) light up.

I have read on the forum that both module ports are planned to be individually powered and could be disabled from code, but I didn't see this functionality exposed yet.

This kind of functionality would come especially handy for LED/Display type modules (to temporarily turn off or reset/power cycle the module without extra wiring etc.).

johnnyman727, 2015-08-10:
+1 I'd like to see this as well.

kevinmehall, 2015-08-10:
The module port power turns on when you open the `/var/run/tessel/port_{a,b}` domain socket and powers off when you close it. `tessel.js` opens both when you `require()` it for T1 compatibility, but with sufficient JS creativity, there could be a way to tell it to open only one. This would also be good for language interoperation -- use one module from JS, one from Rust.

rwaldron, 2015-08-10:
I'll take a crack it.

closed 2015-08-11 by rwaldron