tessel/t2-firmware #83: Couldn't ssh into tessel 2 with existing permissions

kteza1, 2015-08-06:
I'm facing following issue while trying to ssh into tessel after `t2 provision`.

```
ssh -i ~/.tessel/id_rsa root@Tessel-02A320937069.local
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@         WARNING: UNPROTECTED PRIVATE KEY FILE!          @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Permissions 0644 for '/Users/ravitejareddy/.tessel/id_rsa' are too open.
It is required that your private key files are NOT accessible by others.
This private key will be ignored.
bad permissions: ignore key: /Users/ravitejareddy/.tessel/id_rsa
Permission denied (publickey).
```

changing permissions to following fixed this

```
chmod 400 ~/.tessel/id_rsa
```

I'm using Mac OS. I'm not sure if this is relevant but tessel is not connected to my Laptop directly. It is connected to my router though LAN and my Laptop is connected to router through WiFi.

kteza1, 2015-08-06:
UPDATE:

I couldn't reproduce this issue.

I removed existing settings and started over again and I could ssh into tessel without changing permissions

```
rm -rf ~/.tessel/
rm -rf ~/.ssh/known_hosts

t2 provision
ssh -i ~/.tessel/id_rsa root@Tessel-02A320937069.local

```

rwaldron, 2015-08-06:
This bug would actually be in https://github.com/tessel/t2-cli, but was fixed here: https://github.com/tessel/t2-cli/commit/f2c7a46e9cbf763e1af48bb842903da6db041e63 

Files made before that would have the old permissions.

closed 2015-08-06 by rwaldron