tessel/t2-firmware #84: Feature Request: better Neopixel support across GPIO pins

abritinthebay, 2015-08-09:
In t1 the neopixel library is basic... but not bad. However there is a limitation of using only one strip as it is attached to a fixed pin (4 I believe).

Neopixel support on Tessel is great but this is a pretty big limitation. (for example - when running strips that can't realistically be chained together via their data-out-in lines)

johnnyman727, 2015-08-10:
@abritinthebay thanks for the feedback! I've been looking into how we can write the Neopixel driver for T2 and I think it should be no problem to have it work on multiple pins.

abritinthebay, 2015-08-10:
Thanks! Honestly - exposing as much of possible of the NeoPixel api in C to the JS would be great.

Something like the following would be nice: 
```
var hw = process.binding('hw');
var neopixelHW = hw.neopixels;
```
That way whatever abstraction is used the community can improve it/provide alternatives without giving you guys more work!

johnnyman727, 2015-08-18:
That's my plan! Closing this in favor of #24.

closed 2015-08-18 by johnnyman727