tessel/t2-firmware #91: Error on default blinky example from `t2 init`

johnnyman727, 2015-08-21:
I get this error when I run the index.js generated by `t2 init` in the CLI:

```
uffer.js:260
      length += list[i].length;
                       ^
TypeError: Cannot read property 'length' of null
    at Function.Buffer.concat (buffer.js:260:24)
    at null.<anonymous> (/usr/lib/node/tessel-export.js:93:23)
    at emitNone (events.js:67:13)
    at Socket.emit (events.js:166:7)
    at emitReadable_ (_stream_readable.js:408:10)
    at emitReadable (_stream_readable.js:402:7)
    at onEofChunk (_stream_readable.js:387:3)
    at readableAddChunk (_stream_readable.js:129:5)
    at Socket.Readable.push (_stream_readable.js:110:10)
    at Pipe.onread (net.js:552:8)
```

rwaldron, 2015-08-21:
I can take a look this morning

rwaldron, 2015-08-21:
On master: 

```
rwaldron at nova in ~
$ mkdir t2init && cd t2init
rwaldron at nova in ~/t2init
$ t2 init
Initializing tessel repository...
Created package.json.
Wrote 'Hello World' to index.
rwaldron at nova in ~/t2init
$ ls
.
├── [-rw-r--r-- rwaldron staff     405 Aug 21 10:12]  index.js
└── [-rw-r--r-- rwaldron staff     253 Aug 21 10:12]  package.json

0 directories, 2 files
```

Frijol, 2015-08-21:
@rwaldron what does it look like when you `tessel run index.js`?

rwaldron, 2015-08-21:
```
rwaldron at nova in ~
$ mkdir t2init && cd t2init
rwaldron at nova in ~/t2init
$ t2 init
Initializing tessel repository...
Created package.json.
Wrote 'Hello World' to index.
rwaldron at nova in ~/t2init
$ t2 run index.js
INFO Connecting to Tessel...
INFO Connected to arnold over USB
INFO Writing index.js to RAM on arnold (3.072 kB)...
INFO Deployed.
INFO Running index.js...
I'm blinking! (Press CTRL + C to stop)
I'm blinking! (Press CTRL + C to stop)
I'm blinking! (Press CTRL + C to stop)
I'm blinking! (Press CTRL + C to stop)
[...]
^CINFO Stopping script...
```

johnnyman727, 2015-08-22:
@rwaldron is this on `0.0.3`?

Super strange that it runs just fine for you. I'll have to try it again this afternoon.

rwaldron, 2015-08-23:
I can double check tomorrow

johnnyman727, 2015-08-24:
Actually, I can't repro this anymore. Closing for now.

rwaldron, 2015-08-26:
I hit this today and was able to track it and fix it.

johnnyman727, 2015-08-26:
@rwaldron fantastic - how does this error get triggered in the first place?

closed 2015-08-26 by rwaldron

rwaldron, 2015-08-26:
I know you already saw the fix, but for anyone that comes here looking for historic record: https://github.com/tessel/t2-firmware/pull/92/files#diff-84087272b70af9498872cf6f791003abR92