tessel/t2-firmware #107: Add API for Button functionality

HipsterBrown, 2015-10-17:
There is currently no way to access the button input on vessel-2. 
https://tessel.io/docs/hardwareAPI#button-and-leds

What would it take to add this functionality to the `tessel-export.js` module?

tcr, 2015-10-18:
We face a technical question here: Is there an elegant way we can duplicate the GPIO functionality of the ports and extend it to the lone Tessel button?

The Port code is found in https://github.com/tessel/t2-firmware/blob/master/firmware/port.c

The physical button is not part of either port, but all the CMD logic is tied to ports. i figured we can either 

1. duplicate the use of the CMDs to apply to a set of button functions
2. extend the ports to have a “button” pin (which would be the same on either port)

In the former or latter case, we'd want to export the button: https://github.com/tessel/t2-firmware/blob/f88e096fa9dc252bdd9c5404263dd94dddbc9b60/node/tessel-export.js#L48 This would probably look like:

```js
var buttonport = new Tessel.Port('button', Tessel.Port.PATH.button, this)
this.button = new Tessel.Pin(0, buttonport, true, false);
```

A full-stack change, but one which would be easy for us to validate and test.

rwaldron, 2015-10-18:
I had hoped to solve this with a new "reporting button state" byte, so an event could just be emitted, but when designing it, I ran into the "which port emits it" issue.

rwaldron, 2015-10-18:
I'll be available for review all week

kevinmehall, 2015-10-18:
Yeah, that would be my suggestion -- special case the button in the firmware with debouncing in the ISR and send press / release events, so JS doesn't have to deal with the `change` limitation. You could send the event on both ports and make the JS ignore it on port B if both ports are being used.

rwaldron, 2015-10-18:
Or just emit the event from any active Tessel.Port instance and let user code worry about which to concern itself with.

brianarn, 2016-04-14:
For what it's worth, as an excited consumer of my shiny new Tessel 2, who doesn't know enough to make meaningful contributions to the codebase, I'm excited at the idea of being able to play with this button.

I'm hopeful to see some means of using this button soon, and if I knew more about the benefits and risks of the approaches outlined here, I'd try to offer a PR to help it along. Just not sure I'm able to help in that way (... yet?).

johnnyman727, 2016-04-15:
@brianarn thanks for the feedback @brianarn!

This PR would be challenging for a first-timer but not impossible. Here is how I would approach it:

- [ ] add an interrupt handler in the firmware running on SAMD21 for the pin the button is on. You'll want to check out the [schematic](https://s3.amazonaws.com/design-files.tessel.io/2015-06-20/tessel-v2.pdf) to see the pin assignments check out how we already [set interrupts on other pins](https://github.com/tessel/t2-firmware/blob/master/firmware/port.c#L349).
- [ ] When [you get a pin interrupt in JS](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L253), check to see if it was a change on the button, and emit as a `button` event.

GuidoS, 2016-04-30:
@johnnyman727 I just got my tessel2 and was hoping to turn that little button into my internet button sending some rest requests for me. That said, I got as far as "add an" and then I stopped understanding your suggestion.

Can you recommend a learning path to reach an understanding of what you are suggesting. I was excited about tessel because I thought it was possible to build node packages and share them with npm and you are pointing to some c/firmware that is not making me feel very nodely.. ;) Thanks for any help you can share.

johnnyman727, 2016-04-30:
>  I was excited about tessel because I thought it was possible to build node packages and share them with npm and you are pointing to some c/firmware that is not making me feel very nodely.. ;) 

At some level, we have to have compiled code running on a microcontroller :) Once this firmware is written, you would be able to share your Node package with your button code as `tessel.button.on('press', ()=>console.log('so nodely!'))` and everyone would be happy again.

> Can you recommend a learning path to reach an understanding of what you are suggesting.

Yes, potentially. Would you prefer a book suggestion or a long series of programming exercises?

GuidoS, 2016-05-01:
A book would be preferable. Also, seriously, is the button just there for style points??

johnnyman727, 2016-05-01:
I'd recommend reading through Chapter 4 of **Making Embedded Systems** by Elecia White (O'Reilly 2011). By that point, you should understand the general mechanism for pin interrupts in firmware and it won't be much of a stretch to apply it to our codebase.

We have always intended to make the button functional, we've just had higher priorities.

Frijol, 2016-07-11:
Per https://github.com/tessel/project/blob/master/meetings/2016-07-06.md#button-api, the current plan of record is to not expose the button into the API. This lets us use it as a reset button, and encourages users to control their projects through web UI.

sudothinker, 2016-07-11:
While I was hoping for a button API, I really appreciate the communication here and detailed rationale for not supporting it, props to @Frijol  for the meeting notes!

HipsterBrown, 2016-07-19:
This issue is being closed and replaced with 2 issues to support the new functionality:

- https://github.com/tessel/t2-docs/issues/85
- https://github.com/tessel/t2-firmware/issues/185

closed 2016-07-19 by HipsterBrown