tessel/t2-firmware #112: Allow transfers greater than 255 bytes

johnnyman727, 2015-10-18:
Currently, [tessel-export.js](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L387) disallows any transfers more than 255 bytes. This is constrained by the SPI Bridge implementation on the SAMD21.

The easiest fix would be to implement a queue in `tessel-export.js` and keeps track of outgoing/incoming packets. We would only send 255 bytes at at time, wait for the 255 byte response, then send the next 255.

rwaldron, 2015-10-18:
Can you describe the issue that ambient/ir is hitting?

johnnyman727, 2015-10-18:
Yes, Ambient and IR have a dependency on [`avr-isp`](https://github.com/tessel/avr-isp) which is reponsible for updating the module when it has old firmware.

As part of the updating process, it will try to send a multiple kB image over SPI but because that image is over 255 bytes, we get an error:
```
➜  t2-dem  t2 run node_modules/ambient-attx4/examples/ambient.js --name Hangy --lan
INFO Looking for your Tessel...
INFO Connected to Hangy over LAN
INFO Writing examples/ambient.js to RAM on Hangy (275.968 kB)...
INFO Deployed.
INFO Running examples/ambient.js...
Could not retrieve firmware version and module ID. Attempting update.
Uploading new firmware...
/usr/lib/node/tessel-export.js:378
    throw new RangeError('Buffer size must be within 1-255');
    ^

RangeError: Buffer size must be within 1-255
    at Tessel.Port._txrx (/usr/lib/node/tessel-export.js:378:11)
    at Tessel.SPI.transfer (/usr/lib/node/tessel-export.js:725:14)
    at ISP._transfer (/tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:540:12)
    at ISP._flashAll (/tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:275:10)
    at /tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:263:18
    at null.<anonymous> (/tmp/remote-script/node_modules/attiny-common/node_modules/avr-isp/index.js:371:9)
    at null.<anonymous> (/usr/lib/node/tessel-export.js:198:31)
    at emitNone (events.js:67:13)
    at Socket.emit (events.js:166:7)
    at emitReadable_ (_stream_readable.js:411:10)
```

rwaldron, 2015-10-18:
Interesting! Thanks for the details :)

kevinmehall, 2015-10-19:
Support for this on `send` was added here: https://github.com/tessel/t2-firmware/commit/9c1222c45b5febf2764d56cca48f5317b670a323

That would be fairly easy to adapt to use in `_rx` and `_txrx` as well. It doesn't have to wait for the response before sending the next chunk.

I'd also guess that `.send` could be used instead of `transfer`  in the case of the firmware write.

johnnyman727, 2015-10-21:
Thanks, @kevinmehall. Started implementation in #115.

huseyinkozan, 2017-07-24:
#115 is closed, so will this ?

rwaldron, 2017-07-25:
Yep! Thanks again for the follow up

closed 2017-07-25 by rwaldron