tessel/t2-firmware #117: faster boot time

Frijol, 2015-10-26:
It currently takes ~30 seconds for Tessel 2 to boot up. Are there optimizations we can do to make this faster?

cc @kevinmehall for probably having good ideas

kevinmehall, 2015-10-26:
There's a list somewhere of things I was going to do to optimize this. I don't see an issue on openwrt-tessel (firmware boots in milliseconds), so I don't remember where that went. I got the low-hanging fruit, like removing the artificial delays waiting for user input in the bootloader and openwrt failsafe.

The bottleneck is mostly reading data from the SPI flash. Not sure if it will work or will help to just increase the SPI clock speed, but that's a simple tweak worth a try. That's in the devicetree (`.dts`) in the openwrt repo.

If you implemented DMA support for the mt7620 SPI controller in the Linux kernel, that would probably cut boot time in half, but would be rather involved and the mediatek documentation is a little lacking there. Right now the CPU does nothing useful while waiting to read data from flash.

flaki, 2016-04-15:
Whoah linking this in the tessel.io/start is 💯💯!! Transparency FTW! (also, sorry for the off :blush: )

ujjwalt, 2016-11-25:
Has there been an update on this yet?

Frijol, 2016-11-26:
@ujjwalt unfortunately not, but we would love your help! Would you be able to try either of @kevinmehall's suggestions above? Feedback on any attempts (successful or failed) is useful

ujjwalt, 2016-11-26:
I have no knowledge of firmware development or OpenWRT unfortunately. I'm
looking to use Tessel 2 in a product. 30s might be a bummer though every
time. Also is there further documentation on integrating a tessel into a
product?

Frijol, 2016-11-26:
That's something we're working on at the moment– join me on tessel.io/slack --> #learning-wg - I'm happy to discuss

danawoodman, 2018-02-15:
Any updates on this?

JasonKleban, 2020-05-03:
Could something like Yocto Project be used to gen out a custom distro that is faster?  I am completely uninformed on all relevant topics, but since it's quiet in here I don't mind asking.

kevinmehall, 2020-05-06:
@JasonKleban it's mostly a bootloader / kernel driver issue, not anything distro-specific.

mattbasta, 2021-01-23:
Is there a link to a page with more technical information about what's going on here?

andrew-luhring, 2022-04-12:
@kevinmehall @mattbasta any update? i've been trying to get the tessel to work with WSL (windows subsystem for linux) and it hasn't been going super well. i had to install usbipd to connect the subsystems' usb stuff to the actual hardware usb stuff and usbipd is running correctly on the non-tessel device but when i try to mount the tessel it disconnects and reconnects but with an error. i have a feeling it might have something to do with this