tessel/t2-firmware #119: Receiving data over USB before OpenWRT boot causes to usbdaemons to be spawned

johnnyman727, 2015-10-26:
If the CLI tries to send data over USB before OpenWRT is booted, two usbexecd processes are started which seems to break things.

I believe this happens because the coprocessor firmware is forwarding communications on to the SPI Daemon as soon as it boots. The SPI Daemon then receives the communication, notices the USB Daemon hasn't booted and starts it up. Then OpenWRT starts up another USB Daemon as part of its normal boot process.

kevinmehall, 2015-10-26:
There's no code in `spid` to launch `usbexecd`, so that hypothesis isn't correct. If you're seeing multiple `usbexecd` processes in `ps`, my hypothesis is that the extra ones are the child processes that have `fork()`ed but not `exec()`d; that is a child process hanging waiting for command line arguments on the control fd.

johnnyman727, 2015-10-26:
I'll look deeper into that. Thanks @kevinmehall!

johnnyman727, 2015-10-26:
Of course! The CLI tells the daemon to open the process but doesn't ever get the command to send data to the `control` fd. Then the process exits. Then OpenWRT boots up, starts the process, and waits for the `control` stream to have data.

johnnyman727, 2015-10-26:
I do think it makes more sense to block traffic out to OpenWRT from the coprocessor until it's booted.

PR #118 (2015-10-26): Ensures USB comms are not forwarded to OpenWRT before it has booted
Fixes #119.
closed 2015-10-26 by johnnyman727