tessel/t2-firmware #121: DFU update issue

Frijol, 2015-10-27:
Not sure what this is about

```
➜  ~  t2 update -f
INFO Looking for your Tessel...
INFO Connected to Tessel-02A345C70F36 over LAN
INFO Updating Tessel-02A345C70F36 to latest version (0.0.5)...
INFO Beginning update download. This could take a couple minutes..
  downloading [====================] 100% 0.0s remaining
INFO Download complete!
INFO Updating OpenWRT (1/2)
INFO Transferring image of size 20.19 MB
INFO Starting OpenWRT update. Please do not remove power from Tessel.
INFO Updating firwmare (2/2)
Unhandled rejection Error: DFU functional descriptor is invalid
    at new DFU (/usr/local/lib/node_modules/t2-cli/lib/dfu.js:41:15)
    at createDFU (/usr/local/lib/node_modules/t2-cli/lib/usb_connection.js:207:27)
    at tryCatcher (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/util.js:26:23)
    at Promise._settlePromiseFromHandler (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/promise.js:507:31)
    at Promise._settlePromiseAt (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/promise.js:581:18)
    at Promise._settlePromises (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/promise.js:697:14)
    at Async._drainQueue (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/async.js:123:16)
    at Async._drainQueues (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/async.js:133:10)
    at Immediate.Async.drainQueues [as _onImmediate] (/usr/local/lib/node_modules/t2-cli/node_modules/bluebird/js/main/async.js:15:14)
    at processImmediate [as _immediateCallback] (timers.js:368:17)
```

johnnyman727, 2015-10-27:
I got this too! Looks like a regression.

johnnyman727, 2015-10-27:
Yeah, this one is a killer regression. Happens every time.

rwaldron, 2015-10-30:
This is coming from t2-cli: https://github.com/tessel/t2-cli/blob/master/lib/dfu.js#L40-L42

rwaldron, 2015-10-30:
So, this is weird: I hit this error, but the update was successful.

rwaldron, 2015-10-30:
```
$ t2 update
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over LAN
INFO New firmware version found...0.0.5
INFO Updating Tessel-02A36E0C413A to latest version (0.0.5)...
INFO Beginning update download. This could take a couple minutes..
  downloading [====================] 100% 0.0s remaining
INFO Download complete!
INFO Updating OpenWRT (1/2)
INFO Transferring image of size 20.19 MB
INFO Starting OpenWRT update. Please do not remove power from Tessel.
INFO Updating firwmare (2/2)
{ bLength: 9,
  bDescriptorType: 4,
  bInterfaceNumber: 0,
  bAlternateSetting: 0,
  bNumEndpoints: 0,
  bInterfaceClass: 254,
  bInterfaceSubClass: 1,
  bInterfaceProtocol: 2,
  iInterface: 16,
  extra: <Buffer 40 0c 17 10 00 00 00 60 00>,
  endpoints: [] }
Unhandled rejection Error: DFU functional descriptor is invalid
    at new DFU (/Users/rwaldron/clonez/t2-cli/lib/dfu.js:42:15)
    at createDFU (/Users/rwaldron/clonez/t2-cli/lib/usb_connection.js:207:27)
    at tryCatcher (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/util.js:26:23)
    at Promise._settlePromiseFromHandler (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/promise.js:507:31)
    at Promise._settlePromiseAt (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/promise.js:581:18)
    at Promise._settlePromises (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/promise.js:697:14)
    at Async._drainQueue (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/async.js:123:16)
    at Async._drainQueues (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/async.js:133:10)
    at Immediate.Async.drainQueues [as _onImmediate] (/Users/rwaldron/clonez/t2-cli/node_modules/bluebird/js/main/async.js:15:14)
    at processImmediate [as _immediateCallback] (timers.js:368:17)
```

Then...

```
$ t2 update
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over LAN
WARN Tessel-02A36E0C413A is already on the latest firmware version (0.0.5). You can force an update with "t2 update --force".
```

rwaldron, 2015-10-30:
(I added some logging in there)

rwaldron, 2015-10-30:
What is the significance of `0x21` (33)?

rwaldron, 2015-10-30:
Ok, I've run the update 7 times (switching between 0.0.4 and 0.0.5) and it's actually always successful...

```
$ t2 update -f --version=0.0.5

... failure stuff elided

$ t2 version
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over LAN
INFO Tessel [Tessel-02A36E0C413A] version: 0.0.5
$ t2 wifi -n ... -p ...
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over USB
INFO Setting SSID: ... and password: ...
INFO Credentials set!
$ t2 update -f --version=0.0.4

... failure stuff elided

$ t2 version
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over USB
INFO Tessel [Tessel-02A36E0C413A] version: 0.0.4
```

rwaldron, 2015-10-30:
Incidentally, this failure corresponds, time-wise, to the release of node-usb 1.1.0, which was 4 days ago and this appeared 3 days ago.

rwaldron, 2015-10-30:
Confirmed. I just set `nvm use 0.12` and rolled back to `usb@1.0.6`... total success: 

```
$ t2 update -f
INFO Looking for your Tessel...
INFO Connected to Tessel-02A36E0C413A over LAN
INFO Updating Tessel-02A36E0C413A to latest version (0.0.5)...
INFO Beginning update download. This could take a couple minutes..
  downloading [====================] 100% 0.0s remaining
INFO Download complete!
INFO Updating OpenWRT (1/2)
INFO Transferring image of size 20.19 MB
INFO Starting OpenWRT update. Please do not remove power from Tessel.
INFO Updating firwmare (2/2)
INFO Firmware update complete!
INFO Tessel is rebooting. Please wait about 30 seconds before using Tessel.
INFO Force updated Tessel-02A36E0C413A to version 0.0.5
```

rwaldron, 2015-10-30:
@tcr @kevinmehall see previous comment

Frijol, 2015-10-31:
It typically sucessfully updates openWRT but not the firmware, so for the conference we just updated them all until they hit this error, then manually updated firmware.

kevinmehall, 2015-10-31:
The descriptor logged in [rwaldron's comment](https://github.com/tessel/t2-firmware/issues/121#issuecomment-152739756) is a DFU interface descriptor based on the interface class, subclass, protocol, but the DFU Functional descriptor in the `extra` field is garbage. First byte should be the length, second byte should be the DFU descriptor type code (0x21). So the error message is correct, and given that the bootloader hasn't changed and node-usb has, it's a reasonable thing to suspect.

johnnyman727, 2015-10-31:
@kevinmehall you're suggesting that it probably is a regression in `node-usb`?

kevinmehall, 2015-10-31:
Yes, and I think I found it: https://github.com/nonolith/node-usb/pull/100

johnnyman727, 2015-11-14:
Confirmed that nonolith/node-usb#100 is the fix! We'll need to bump the version in package.json after a new node-usb version is released.

rwaldron, 2015-11-14:
If you see this first, nevermind my question on the other issue ;)

closed 2015-12-03 by johnnyman727