tessel/t2-firmware #122: Python flash script has broken

johnnyman727, 2015-10-27:
I have a Tessel with a corrupted OpenWRT image so I was going to flash it over USB using `flash.py` in `/scripts`. I have a folder with the proper uboot and squashfs binary image. However, the script consistently fails on writing the squashfs image with a timeout error (I've attempted to increase the failing `read` with a timeout of 1s with no luck):

```
➜  scripts git:(master) ✗ python flash.py ~/Downloads/new_build/
Generated MAC addr  02:a3:9b:5c:75:d3
 FInally Here
Traceback (most recent call last):
  File "flash.py", line 144, in <module>
    flash.write_tessel_flash(basepath, mac1, mac2)
  File "flash.py", line 123, in write_tessel_flash
    self.write(0x50000, open(os.path.join(path, 'openwrt-ramips-mt7620-tessel-squashfs-sysupgrade.bin')).read())
  File "flash.py", line 110, in write
    self.write_page(write_addr, map(ord, p))
  File "flash.py", line 101, in write_page
    self.transaction([0x12]+address(addr)+data, status_poll = True, wren = True)
  File "flash.py", line 42, in transaction
    self.ep_out.write(hdr + write)
  File "/Library/Python/2.7/site-packages/pyusb-1.0.0b2-py2.7.egg/usb/core.py", line 349, in write
  File "/Library/Python/2.7/site-packages/pyusb-1.0.0b2-py2.7.egg/usb/core.py", line 878, in write
  File "/Library/Python/2.7/site-packages/pyusb-1.0.0b2-py2.7.egg/usb/backend/libusb1.py", line 760, in bulk_write
  File "/Library/Python/2.7/site-packages/pyusb-1.0.0b2-py2.7.egg/usb/backend/libusb1.py", line 856, in __write
  File "/Library/Python/2.7/site-packages/pyusb-1.0.0b2-py2.7.egg/usb/backend/libusb1.py", line 552, in _check
usb.core.USBError: [Errno 60] Operation timed out
```

@kevinmehall do you think this could be an issue with the addresses in flash where the squashfs image is being written? Perhaps it has changed?

johnnyman727, 2015-12-17:
Current workaround (from within the root directory of this repo):
```
# checkout the last known working version of t2-firmware with updating script
git checkout 7af3b341043dffd4fa78704feff033bb23bdd341;
# rebuild the firmware
make clean; make;
# put device into bootloader mode by holding down button when plugging in
# then flash the old firmware onto the device
dfu-util -aFlash -d 1209:7551 -D build/firmware.bin
# Put the boot image and squashfs image into a folder. You can use [this example](https://github.com/tessel/t2-firmware/files/65971/update-openwrt.zip) but make sure you unzip it first
python flash.py PATH_TO_THE_FOLDER_WITH_IMAGES

# you will see the following output:
Chip id 010219
Chip erase...
Chip erase complete (57.05s)
Write 0x00000000 (100%, 0.25s)
Generated MAC addr  02:a3:cd:4d:3f:32
Write 0x00040000 (100%, 0.00s)
Write 0x00050000 (100%, 54.87s)

# update is complete. checkout latest firmware
git checkout master
# rebuild the firmware
make clean; make;
# put device into bootloader mode by holding down button when plugging in
# then flash the current firmware onto the device
dfu-util -aFlash -d 1209:7551 -D build/firmware.bin

johnnyman727, 2015-12-17:
A `git bisect` reveals that the culprit is this commit: https://github.com/tessel/t2-firmware/commit/690ca5786521beb8c28c4b1df68f01f00e2d4e0c

johnnyman727, 2015-12-17:
Commenting out the [systick initialization](https://github.com/tessel/t2-firmware/commit/690ca5786521beb8c28c4b1df68f01f00e2d4e0c#diff-f5275649d3fca868896862b40f4f317cR138) seems to solve the problem but I'm not yet sure why that would interfere with Flash writing. Perhaps the systick handler is interrupting SPI writes.

reconbot, 2015-12-17:
What does `Put the boot image and squashfs image into a folder` mean?

rwaldron, 2015-12-17:
```
$ python flash.py update-openwrt/
Traceback (most recent call last):
  File "flash.py", line 104, in <module>
    chip_id = rdid()
  File "flash.py", line 38, in rdid
    return transaction([0x9f], 3)
  File "flash.py", line 26, in transaction
    dev.write(0x02, [(read >> 0) & 0xff, (read >> 8) & 0xff, (read >> 16) & 0xff, flags] + write)
  File "/Library/Python/2.7/site-packages/usb/core.py", line 611, in write
    fn = fn_map[self._ctx.get_endpoint_type(self, endpoint, intf)]
  File "/Library/Python/2.7/site-packages/usb/core.py", line 181, in get_endpoint_type
    etype = util.endpoint_type(e.bmAttributes)
AttributeError: 'NoneType' object has no attribute 'bmAttributes'
```

johnnyman727, 2015-12-17:
@reconbot I'm not sure of a more concise way to describe it (mostly because I don't understand OpenWRT all that well myself). But the script expects a bootloader binary and a squashfs (the main file system) binary to be in the same folder. If you download the file I uploaded and extract it, you'll see what I mean.

@rwaldron I wonder if that's a Python version issue. My `python --version` returns 2.7.9.

johnnyman727, 2015-12-17:
or an older version of `usb core`? This is why I hate Python.

rwaldron, 2015-12-17:
Updated to python 2.7.9, same error. How do I update `usb core`?

johnnyman727, 2015-12-17:
Not sure - I think that it should be updated as you update Python versions.

Just to be clear, the device isn't in bootloader mode, right?

johnnyman727, 2015-12-17:
Can you add a ` print(dev)` below where `dev` is instantiated? (just below line 126)

rwaldron, 2015-12-17:
There isn't a line 126 in that script for the sha you mentioned above

rwaldron, 2015-12-17:
```python
from __future__ import print_function
import usb.core
import random
import sys
import os
import time

dev = usb.core.find(idVendor=0x9999, idProduct=0xffff)
if dev is None:
    raise ValueError('Our device is not connected')

def chunks(l, n):
    for i in xrange(0, len(l), n):
        yield l[i:i+n]

dev.set_interface_altsetting(interface = 0, alternate_setting = 1)

def address(addr):
    return [(addr>>24) & 0xFF, (addr>>16) & 0xFF, (addr>>8) & 0xFF, addr & 0xFF]

def transaction(write, read=0, status_poll=False, wren=False):
    if len(write) > 500 or read >= 2**24:
        raise ValueError("Transaction too large")

    flags = int(status_poll) | (int(wren) << 1)
    dev.write(0x02, [(read >> 0) & 0xff, (read >> 8) & 0xff, (read >> 16) & 0xff, flags] + write)

    if read > 0:
        return bytearray(dev.read(0x81, max(read, 512)))
    else:
        return bytearray()

def showhex(buf):
    print(''.join('{:02x}'.format(x) for x in buf))

def rdid():
    return transaction([0x9f], 3)

def read(addr, length):
    return transaction([0x13]+address(addr), length)

def wren():
    transaction([0x06])

def status():
    return transaction([0x05], 1)[0]

def wait_complete():
    count = 0
    while True:
        count += 1
        s = status()
        if s & 1 == 0:
            break
        if count > 50:
            time.sleep(0.1)

def erase_sector(addr):
    wren()
    print("Erase {:x}".format(addr))
    transaction([0xDC]+address(addr))
    wait_complete()

def chip_erase():
    wren()
    print("Chip erase...")
    t = time.time()
    transaction([0x60])
    wait_complete()
    print("Chip erase complete ({:.2f}s)".format(time.time()-t))

def write_page(addr, data):
    transaction([0x12]+address(addr)+data, status_poll = True, wren = True)

PAGE = 256

def write(addr, data):
    t = time.time()
    write_addr = addr
    for i, p in enumerate(chunks(data, PAGE)):
        if i % 32 == 0:
            print("Write 0x{:08x} ({:3.0f}%)\r".format(addr, i*PAGE*100.0/ len(data)), end='')
        write_page(write_addr, map(ord, p))
        write_addr += PAGE
    wait_complete()
    print("\rWrite 0x{:08x} (100%, {:.2f}s)".format(addr, time.time() - t))

def randbyte():
    return random.randint(0, 255)

def factory():
    header = [0x20, 0x76, 0x03, 0x01]
    uid = [randbyte(), randbyte(), randbyte(), randbyte()]
    mac1 = [0x02, 0xa3] + uid
    mac2 = [0x02, 0xa4] + uid

    print("Generated MAC addr ", ':'.join("{:02x}".format(x) for x in mac1))
    return ''.join(map(chr, header + mac1 + [0xff] * 30 + mac2))

basepath = sys.argv[1]
assert(basepath)

chip_id = rdid()
print("Chip id", str(chip_id).encode('hex'))

if chip_id != bytearray([0x01, 0x02, 0x19]):
    print("Invalid chip ID (flash communication error)")
    exit(1)

t = transaction([0x05] + [0]*100, 1000)
assert(len(t) == 1000)

chip_erase()

write(0,       open(os.path.join(basepath, 'openwrt-ramips-mt7620-Default-u-boot.bin')).read())
write(0x40000, factory())
write(0x50000, open(os.path.join(basepath, 'openwrt-ramips-mt7620-tessel-squashfs-sysupgrade.bin')).read())
```

johnnyman727, 2015-12-17:
sorry, was looking at different rev (but you're on the correct on). It would be after line 8 in this case.

rwaldron, 2015-12-17:
```
$ python flash.py update-openwrt/
<usb.core.Device object at 0x101689810>
Traceback (most recent call last):
  File "flash.py", line 105, in <module>
    chip_id = rdid()
  File "flash.py", line 39, in rdid
    return transaction([0x9f], 3)
  File "flash.py", line 27, in transaction
    dev.write(0x02, [(read >> 0) & 0xff, (read >> 8) & 0xff, (read >> 16) & 0xff, flags] + write)
  File "/Library/Python/2.7/site-packages/usb/core.py", line 611, in write
    fn = fn_map[self._ctx.get_endpoint_type(self, endpoint, intf)]
  File "/Library/Python/2.7/site-packages/usb/core.py", line 181, in get_endpoint_type
    etype = util.endpoint_type(e.bmAttributes)
```

johnnyman727, 2015-12-17:
Do you have other USB devices connected? I notice that the vendor and product id it's looking for are weird (0x9999 and 0xffff). Try using the version of this script at `4b2f05e2bf97c6672fcc33b77846cb709a1655e3` (it uses the correct vID and pID)

rwaldron, 2015-12-17:
```
$ python flash.py update-openwrt/
Traceback (most recent call last):
  File "flash.py", line 11, in <module>
    raise ValueError('Our device is not connected')
ValueError: Our device is not connected
```

rwaldron, 2015-12-17:
I have to head out now

johnnyman727, 2015-12-17:
Okay.

When you get a chance to work on this again, be sure to run `make;` and the `dfu-util` command to flash the device before trying the script again.

rwaldron, 2015-12-17:
Still the same

```
$ python flash.py update-openwrt/
Traceback (most recent call last):
  File "flash.py", line 11, in <module>
    raise ValueError('Our device is not connected')
ValueError: Our device is not connected
```

I will try all of this from the top tomorrow

johnnyman727, 2015-12-22:
Fixed in #141.

closed 2015-12-22 by johnnyman727

rwaldron, 2015-12-23:
- Updated master
- `make clean; make`
- Downloaded https://builds.tessel.io/t2/firmware/d34f30a766081f11b46011c802ce64fe6559755a.tar.gz and moved openwrt.bin into `build/`
- Result: 
```
rwaldron at nova in ~/clonez/t2-firmware on master*
$ python scripts/flash.py build/
Traceback (most recent call last):
  File "scripts/flash.py", line 128, in <module>
    raise ValueError('Our device is not connected')
ValueError: Our device is not connected
```

rwaldron, 2015-12-23:
> The good news is that I discovered what appears to be a Tessel that's functional.

About that...

```
root@ash:~# cat /etc/tessel-version
1293f03d42ae00b11267e5af2d6739c8cac77165
```

Which corresponds to 0.0.5 :\

rwaldron, 2015-12-23:
UPDATE. 

After I went through all of this, I figured one more try can't hurt. Here are the results...

- `bishop`
  - Powered on, boot indicator actually _finished_. 
  - `t2 list` cannot find it
  - I _can_ Serial into the board

    ```
    Tessel 2  /  Built on OpenWrt
    root@Tessel-02A3833F9836:/# cat /etc/tessel-version 
    d34f30a766081f11b46011c802ce64fe6559755a
    ```
      - `d34f30a766081f11b46011c802ce64fe6559755a` is 0.0.7
      - Note that it's lost it's name, I only call it `bishop` because it has a brother p-touch label with that name ;)

  - While back in `t2-firmware` (with the latest master build), I unplugged the device-formerly-known-as-bishop and put it into bootloader mode.
    - `dfu-util -aFlash -d 1209:7551 -D build/firmware.bin` finished successfully.
    - disconnect & reconnect; boot indicator finished, again! And then...

      ```
      rwaldron at nova in ~/clonez/t2-firmware on master*
      $ t2 list
      INFO Searching for nearby Tessels...
        Tessel-02A3833F9836 USB
      rwaldron at nova in ~/clonez/t2-firmware on master*
      $ t2 wifi -n JLY4Q -p XCCGM3C43SC7JVDX
      INFO Looking for your Tessel...
      INFO Connected to Tessel-02A3833F9836.
      INFO Successfully connected!
      INFO Wifi Enabled.
      INFO Wifi connection successful. SSID: JLY4Q, password XCCGM3C43SC7JVDX, security mode: psk2
      ```    

Recovered!

I will attempt the same process with the others

johnnyman727, 2015-12-23:
@rwaldron this is pretty surprising. The last logs for `bishop` showed that its Flash was completely erased and it errored before any of the OpenWRT image was flashed onto it. Did you flash it again?

rwaldron, 2015-12-23:
- `maria` never finishes booting. It's been "booting" since I posted that last update.

rwaldron, 2015-12-23:
> Did you flash it again?

No, every step I took is written here.