tessel/t2-firmware #125: can't create new UART devices once one has been created...

johnnyman727, 2015-11-23:
On Tessel 1, you could switch UART parameters by simply creating a new device:
```
  //  First, I want a faster baud
  this.uart = this.hardware.UART({baudrate : 115200});
  //  Them, I want to reset UART baud rate to 9600
  this.uart = this.hardware.UART({baudrate : 9600});
```
This GPS module driver does this because the GPS wants to speak 115200 initially but then we need to tell it to speak at 9600.

But you can't do this currently on Tessel 2 because we check to see if a [UART interface has already been made](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L270-L276), then we return that existing port.

johnnyman727, 2015-11-23:
I'm not sure what the best practice would be here. Perhaps from here on out we want to encourage developers to call `this.uart.disable()` before creating another interface?

rwaldron, 2015-11-23:
Or just auto-disable the existing one when a new one is created?

johnnyman727, 2015-11-24:
@rwaldron yeah, that's a better option. Updated my PR - it will automatically disable old interface if a new one is requested.

PR #126 (2015-11-24): Allow new UART and SPI interface creations
Can potentially fix #125.
closed 2015-11-24 by johnnyman727