tessel/t2-firmware #129: [RFC] Expose Tessel network utilities

HipsterBrown, 2015-12-01:
At the moment, the quickest way to connect to a wifi network or start an access point is through the command line interface, `t2-cli`. It would be cool to expose this functionality through the `tessel` module for controlling network operations programatically. 

Example:
```js
var tessel = require('tessel');

// connects to wifi network called 'MyNetwork' and returns wifi object for use later
// tessel.network.ap could work the same way
var myNetwork = tessel.network.wifi({
  ssid: 'MyNetwork',
  password: 'SecurePassword123',
  security: 'psk2'
});

// turns off the connection to 'MyNetwork'
myNetwork.off();

// turns on connection to 'MyNetwork'
myNetwork.on();
```

Use Case:
If a developer wanted to create an access point based on user interaction, like pressing the button on the Tessel. Then a user could toggle that access point every time they press the button. 

When mesh networking is available, that could also be exposed through this `tessel.network` object/class  for dynamically creating mesh nodes and connections. 

Any feedback is helpful. I can work on a PR if there is a general approval.

rwaldron, 2015-12-01:
What happens if `tessel.network.wifi` is called more than once?

HipsterBrown, 2015-12-01:
Hmmm I suppose it would connect to that new network and return a new wifi object, then invalidate any old wifi objects. I think having that returned object to pass around will be easier to understand, as opposed to:

```js
var tessel = require('tessel')

tessel.network.wifi(creds); automatically connects to this network
tessel.network.wifi.off(); // turns the last connected network off.
```

johnnyman727, 2015-12-02:
@HipsterBrown what do you think of [the WiFi API we have for Tessel 1](https://github.com/tessel/t1-docs/blob/master/wifi.md#connecting-to-wifi-from-js)? It doesn't have any access point option so we'd have to add that.

HipsterBrown, 2015-12-02:
@johnnyman727 I was not aware of that API but it looks great to me, especially since it existed with Tessel 1 it would be nice to continue support within the `tessel` module. I think the same API should work for access points as well.

johnnyman727, 2015-12-02:
:+1: Yeah, I forgot about this API so I'm glad you brought it up.

rwaldron, 2015-12-03:
I **also** forgot about that API—thanks for the refresher @johnnyman727

johnnyman727, 2015-12-22:
@HipsterBrown do you think this will be ready for our next milestone tomorrow? Are you blocked by anything?

HipsterBrown, 2015-12-22:
@johnnyman727 It won't be ready tomorrow. Not blocked by anything, just haven't had a bunch of time to work on it. I can have it by the next release though now that I have some feedback on the initial WIP API.

johnnyman727, 2015-12-22:
:+1: Will move it to the next milestone.

PR #140 (2016-02-10): Network API
Fixes #129 

This build covers a proposed Network Wifi API. The API is inspired by https://github.com/tessel/t1-docs/blob/master/wifi.md#connecting-to-wifi-from-js .

Example:

```js
var tessel = require('tessel');

// when "connect" event fires, log the returned settings
tessel.network.wifi.on('connect', console.log.bind(console));

// connect to wifi network
tessel.network.wifi.connect({
  ssid: 'MyNetwork',
  password: 'MyPassword',
  security: 'psk2'
}, function(error, settings) {
  if (error) {
    throw error;
  }
  
  // logs the settings and network info returned, same info [... cut]
closed 2016-02-10 by HipsterBrown