tessel/t2-firmware #135: Make ethernet connections work out of the box

johnnyman727, 2015-12-09:
I'd like to be able to plug in an ethernet port and be able to use Tessel's LAN connection, even when not using wifi. Should also be able to send data to/from the internet. I believe this entails a modification to `/etc/config/network` somehow but I don't know the specifics.

HipsterBrown, 2015-12-09:
The internet is a big, scary place; are you sure you want communicate with it? ;)

So you want to have a connection available as soon as you plug in an ethernet cable into the port? Without running any command in the cli?

I wonder if there is some hardware event to listen to when an ethernet connection is available. 

Also, does it not connect automatically after plugging in an ethernet cable?

johnnyman727, 2015-12-09:
> The internet is a big, scary place; are you sure you want communicate with it? ;)

I'm not sure I want to but I'm sure I want the option to :)

> So you want to have a connection available as soon as you plug in an ethernet cable into the port? Without running any command in the cli?

Yes, that's what I personally would expect to happen. I mean, perhaps we also provide the option to enable/disable from the CLI but I think the physical act of plugging in the connection is a good indicator that the user would like to use it.

> I wonder if there is some hardware event to listen to when an ethernet connection is available.

I would bet that `ubus` emits something of the sort if `/etc/config/network` and friends are configured properly.

> Also, does it not connect automatically after plugging in an ethernet cable?

I'm not a networking expert by any means but I just plugged my Tessel's ethernet port into one of the open ports on my router and was not able to ping it, discover it over mDNS, etc.

HipsterBrown, 2015-12-09:
> I just plugged my Tessel's ethernet port into one of the open ports on my router and was not able to ping it, discover it over mDNS, etc

Ah ok. Can you `dterm` in while it's connected to the router to see if it's getting a network connection but it's not available via LAN discovery for some reason? 

Looking through the OpenWRT docs, I haven't seen much about ethernet and the config involved in that. But there may be something I'm missing.

johnnyman727, 2015-12-09:
> Ah ok. Can you dterm in while it's connected to the router to see if it's getting a network connection but it's not available via LAN discovery for some reason?

I wish so badly that I could. But, unfortunately, I upgraded to OSX El Capitan because 'why not' which apparently has *no* support for serial terminals and there is no way to revert back to a previous version. I'm been living a `dterm`-less life for about a month now and it's pretty tough :sob:

HipsterBrown, 2015-12-09:
Dang, my work computer is still on Yosemite but I don't have an ethernet adapter or extra cord at home right now. I can try it out this weekend though. The lack of `dterm` has been the biggest pain about El Cap so far. :/

HipsterBrown, 2015-12-12:
So I recently found out that devices connected to a router are not discoverable over LAN. I suppose it could be a security measure to not expose Ethernet-connected devices. ¯\_(ツ)_/¯

reconbot, 2015-12-12:
What LAN/router? That's very odd to me. They should be discoverable if
they're on the same network.

HipsterBrown, 2015-12-13:
It may only be true in the case I heard it. Someone was telling me about how they're Apple TV was not discoverable over AirPlay when plugged into the Ethernet, but that could be a bug in the AirPlay or Apple TV source.

reconbot, 2015-12-14:
Some routers are known to be pains.

johnnyman727, 2016-02-08:
I retried this today and had a working ethernet connection. I'm not sure what changed but I'm going to close this issue for now. Feel free to re-open if anyone encounters this issue again.

Smoke Test:
* `t2 list` (ensure you can see the LAN connection to the device)
* `ping google.com` (from the T2 shell, ensure you get responses)
* `ping 192.168.8.102` (from the T2 shell, pinging my host computer, ensure you get responses)

closed 2016-02-08 by johnnyman727