tessel/t2-firmware #142: DFU Bootloader Split Out

peter-conalgo, 2015-12-19:
I plan to take your wonderful USB DFU Bootstrap code and pull it out into a new stand-alone project.

My goals:
- generalize it so can run on any D21 (and maybe others) ideally without pre-configuration
- make it easy to rebrand and/or be generic
- minimalist and self-contained Makefile
- make it easy to link into other projects as a linkable object
- compile-time option for a secure mode that blocks any reads
- support pre and post boot customization options
- stay under 4k and open-source

I am doing this because I need a bootloader for an open source project I'm just starting, plus I might use in it in commercial projects. 

My questions:
- what should I use for copyright notices and licenses?

johnnyman727, 2015-12-19:
Hi @peter-conalgo thanks for getting in touch. This is a good question for @kevinmehall who wrote most, if not all, of the USB code.

peter-conalgo, 2015-12-20:
Project has been named "DAFU".

Please see new repo at: <https://github.com/opendime/DAFU>

johnnyman727, 2015-12-20:
Fantastic - let me know if you have any questions about the code.

johnnyman727, 2016-02-05:
@peter-conalgo how are things going with this project?

peter-conalgo, 2016-02-05:
Done and stable as far as I'm concerned. I'm using it with success on two projects.

johnnyman727, 2016-02-05:
Fantastic! I'm going to go ahead and close this issue but feel free to reach out if we can help in any way.

closed 2016-02-05 by johnnyman727