tessel/t2-firmware #152: pin.on 'change' event not working properly

io2work, 2016-03-01:
I am playing with the `pin.pull` right now and I am still experiencing some odd behavior. If I run a simple `setInterval` loop like the following it works fine... meaning I have the `pin.pull('pulldown')` set and when I push a button it sends 3.3V to that pin and will toggle back and forth between 0 and 1 as expected.

```
setInterval(function() {
    pin.read(function(err, val) {
        console.log('pull down val', val);
    });
 }, 500);
```

But, if I setup a `change` event on the pin it will only read the event the first time the button is pressed (it actually reads it twice for some reason) but it never returns to 0 an therefore the `change` event never fires again.

```
pin.on('change', function() {
    pin.read(function(err, val) {
        console.log('pull down val', val);
    });
});
```

dudleyjosh, 2016-03-25:
@rwaldron I have tried to debug this on my own without luck yet... do you have any idea what might be causing this behavior?

rwaldron, 2016-03-25:
Sorry, it seems I missed this when you first posted. Will try to repro now

rwaldron, 2016-03-25:
Ok, I put together this circuit: 

[image]

And ran this code: 

```js
[2, 5, 6, 7].forEach(index => {
  tessel.ports.A.pin[index].on('change', function() {
    console.log(index);
  });
});
```

And I'm able to observe change events for the life of the program. What I find problematic is that I don't know the state of the pin when that change event is fired. I'm going to attempt to co-opt bit 3 as a "value field", ie. 

With a `console.log(replyBuf);` on line 173 of tessel-export.js. 

| Pin   | Byte    | Bits: As-is    | Proposed Bits: High |
| ----- | ------- | ------------- | ------------------- |
| 2     | `0xc2`  | `0b11000010`  | `0b11001010`        |
| 5     | `0xc5`  | `0b11000101`  | `0b11001101`        |
| 6     | `0xc6`  | `0b11000110`  | `0b11001110`        |
| 7     | `0xc7`  | `0b11000111`  | `0b11001111`        |

dudleyjosh, 2016-03-25:
I wasn't using the resistor because I was using the built-in pulldown/pullup resistors running the code below. I just re-tested it based on what you sent and it seems to work fine without the read() within the on-change function.

```javascript
var tessel = require('tessel'); // import tessel

var pin = tessel.port['B'].pin[2];
pin.pull('pulldown');

pin.on('change', function() {
    pin.read(function(err, val) {
        console.log('val', val);
    });
});
```

rwaldron, 2016-03-25:
Cool. I also think it's important that you should _never_ have to _also_ ask for the value when you get a change event... so I implemented my idea above and it works exactly as I'd hoped. Bit 3 of the interrupt response byte was not in use, bits 0, 1 & 2 were used to indicate that a "change" occurred on a "pin". You could know what pin, but not what its state was... so I put the state in bit 3 and it works perfectly. Patch and tests to follow.

dudleyjosh, 2016-03-25:
This will definitely work, but I still don't understand why the read() within the change event would cause the problem I was seeing? I'm just curious as to what was happening I guess.

dudleyjosh, 2016-03-28:
I played with this some more and discovered that it works fine with `.rawRead()` but `.read()` definitely doesn't work when called within the callback of the `.on('change')` event.

```javascript
var tessel = require('tessel');
var tesselPort = 'B';

var changeCount = 0;

var pins = [2, 5, 6, 7];

// works with .rawRead() but not with .read()
pins.forEach(pinNum => {
    tessel.port[tesselPort].pin[pinNum].pull('pulldown');
    tessel.port[tesselPort].pin[pinNum].on('change', function() {
        changeCount++;
        console.log('pin:', pinNum, 'changeCount:', changeCount);

        // this works
        this.rawRead(function(err, val) {
            console.log('val:', val);
        });

        // this doesn't work
        /*
        this.read(function(err, val) {
            console.log('val:', val);
        });
        */

    });
});

setImmediate(function() {
    console.log('ready');
});
```

closed 2016-04-02 by johnnyman727