tessel/t2-firmware #162: Blinky.js from examples is broken on Tessel 2.

chaplinkyle, 2016-04-03:
Blinky.js from http://start.tessel.io/blinky is broken. Undefined is returned when accessing the led because output() does not return 'this'.

HipsterBrown, 2016-04-03:
@chaplinkyle Thanks for making an issue for this. Those docs are specific to the Tessel 1. 

The blinky.js example for the Tessel 2 is available here -> http://tessel.github.io/t2-start/blinky.html

Could you confirm that the Tessel 2 example works for you?

chaplinkyle, 2016-04-03:
That works great. Thanks!. I suggest still merging my pull request though for the sake of backwards compatibility. Especially since the comment about the function is '// Define backward compatibility alias'

johnnyman727, 2016-04-04:
Thanks for submitting this @chaplinkyle. 

The reason we didn't return `this` when we made the API (and I'm happy to continue discussing if you disagree) is because hardware operations are now asynchronous. We have the microprocessor sending commands (eg. pull this pin high) to the microcontroller and the microcontroller reporting back the success to the microprocessor. We thought it could be dangerous to return `this` because it encourages chaining. Chaining may not work out exactly how you expect when the commands are asynchronous (for example, if one command has an error, the other commands will still be queued which can cause more errors) so it seems better to remove that capability altogether.

chaplinkyle, 2016-04-04:
Sounds good. Thanks for the explanation.

closed 2016-04-04 by chaplinkyle