tessel/t2-firmware #164: pin.on('change', function(pinVal){}) not always providing pinVal properly

dudleyjosh, 2016-04-08:


rwaldron, 2016-04-08:
To clarify a bit about this: the issue doesn't appear until a user program tries to observe more than one interrupt pin (set to input). For example: 

```js
var tessel = require('tessel');
var pinMode = 'pulldown';

tessel.ports.A.pin[2].pull(pinMode);
tessel.ports.A.pin[2].on('change', function(pinVal) {
  console.log(pinVal);
});
```

Is fine: 

```
$ t2 run index.js
INFO Looking for your Tessel...
INFO Connected to bishop.
INFO Building project.
WARN The following items were found in your project directory, but not deployed.
INFO Writing project to RAM on bishop (3.072 kB)...
INFO Deployed.
INFO Running index.js...
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
...
```

And...

```js
var tessel = require('tessel');
var pinMode = 'pulldown';

[2, 5].forEach(function(pin) {
  tessel.ports.A.pin[pin].pull(pinMode);
  tessel.ports.A.pin[pin].on('change', function(pinVal) {
      console.log(pin, pinVal);
  });
});
```

Is fine...

```
$ t2 run index.js
INFO Looking for your Tessel...
INFO Connected to bishop.
INFO Building project.
INFO Writing project to RAM on bishop (3.072 kB)...
INFO Deployed.
INFO Running index.js...
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
```

But, then...

```js
var tessel = require('tessel');

[2, 5].forEach(function(pin) {
  tessel.ports.A.pin[pin].input();
  tessel.ports.A.pin[pin].on('change', function(pinVal) {
      console.log(pin, pinVal);
  });
});
```

Has the following result: 

```
$ t2 run index.js
INFO Looking for your Tessel...
INFO Connected to bishop.
INFO Building project.
INFO Writing project to RAM on bishop (3.072 kB)...
INFO Deployed.
INFO Running index.js...
2 0
5 0
2 0
5 0
2 1
5 1
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
```

dudleyjosh, 2016-04-08:
Something else odd is going on here too? Running the following code:

```javascript
[2, 5].forEach(function(pin) {
    tessel.ports.A.pin[pin].pull(pinMode);
    tessel.ports.A.pin[pin].on('change', function(pinVal) {
        console.log(pin, pinVal);
    });
});
```

Console with both pins connected to the button:

```
$ t2 run pintest.js
INFO Looking for your Tessel...
INFO Connected to rde.
INFO Building project.
INFO Writing project to RAM on rde (5.12 kB)...
INFO Deployed.
INFO Running pintest.js...
2 1
2 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
2 1
5 1
2 0
5 0
```

With pin 5 disconnected:
```
2 0
2 0
2 0
2 0
2 0
2 0
2 0
2 0
2 0
```

With pin 2 disconnected:
```
5 1
5 1
5 0
5 1
5 1
5 0
5 1
5 0
5 1
5 0
```