tessel/t2-firmware #166: Pin.analogRead intermittently unresponsive

jziggas, 2016-04-11:
[Forum post reference](https://forums.tessel.io/t/pin-analogread-intermittently-unresponsive/2330)

I'm trying to call analogRead a couple hundred times and sometimes it gets hung up. I'm pretty new to hobby electronics but is this something I should call "serially" rather than in "parallel"? It seems to get hung up either way.

For example if you try to run:

```
for (var i = 0; i < 200; i++) {
    pin.analogRead(function(error, value) {
        // do something
    });
}
```
It definitely bonks out. But if instead I try to call analogRead "serially" by waiting for the previous callback to be invoked before reading the pin again, I still get the occasional instance where a few readings are taken and then the app just hangs after that. Curious if anyone else has experienced this (or even tried it yet).

johnnyman727, 2016-04-12:
Thanks @jziggas. Will look into this soon.

dudleyjosh, 2016-04-12:
I just gave this a try (running v0.0.12 firmware and using v0.0.17 CLI) and it seemed to work ok?

```
var tessel = require('tessel');
var pin = tessel.ports.A.pin[7];

var count = 0;

for (var i = 0; i < 200; i++) {
    pin.analogRead(function(error, value) {
        count++;
        console.log(count, value);
    });
}
```
output:
```
1 2.8866943359375
2 2.2953369140624997
3 1.9746826171874998
4 1.8385253906249999
5 1.7394287109375
6 1.6765869140624998
7 1.6435546875
8 1.6072998046875
9 1.5847412109374999
10 1.5799072265624998
11 1.5726562499999999
12 1.554931640625
13 1.5533203125
14 1.558154296875
15 1.546875
16 1.5492919921875
17 1.5541259765625
18 1.554931640625
19 1.5460693359375
20 1.542041015625
21 1.5323730468749999
22 1.5299560546875
23 1.542041015625
24 1.5315673828124998
25 1.5444580078124999
26 1.5476806640624998
27 1.5444580078124999
28 1.5412353515625
29 1.55009765625
30 1.5533203125
31 1.5492919921875
32 1.5533203125
33 1.5492919921875
34 1.5476806640624998
35 1.5057861328125
36 1.506591796875
37 1.4920898437499999
38 1.4945068359375
39 1.4783935546875
40 1.4558349609374999
41 1.4364990234375
42 1.4373046875
43 1.4558349609374999
44 1.442138671875
45 1.2737548828124998
46 1.2995361328125
47 1.34384765625
48 1.377685546875
49 1.4155517578124999
50 1.4091064453125
51 1.4074951171874999
52 1.416357421875
53 1.429248046875
54 1.264892578125
55 1.329345703125
56 1.368017578125
57 1.3704345703124998
58 1.3905761718749998
59 1.42119140625
60 1.426025390625
61 1.4526123046875
62 1.432470703125
63 1.4461669921875
64 1.4228027343749998
65 1.3946044921874998
66 1.3978271484375
67 1.4034667968749999
68 1.4123291015625
69 1.4308593749999998
70 1.442138671875
71 1.45341796875
72 1.4284423828125
73 1.4074951171874999
74 1.397021484375
75 1.4203857421875
76 1.4655029296875
77 1.490478515625
78 1.4896728515625
79 1.4574462890625
80 1.448583984375
81 1.426025390625
82 1.2592529296875
83 1.2906738281249999
84 1.3446533203125
85 1.3591552734374999
86 1.3712402343749999
87 1.4074951171874999
88 1.4091064453125
89 1.405078125
90 1.4236083984374999
91 1.4622802734375
92 1.4993408203124998
93 1.4598632812499999
94 1.4791992187499998
95 1.2979248046874998
96 1.3672119140624999
97 1.429248046875
98 1.4550292968749998
99 1.2785888671874999
100 1.3559326171875
101 1.4219970703125
102 1.4638916015624999
103 1.480810546875
104 1.5017578125
105 1.513037109375
106 1.3140380859374998
107 1.3897705078125
108 1.4397216796874999
109 1.4679199218749999
110 1.4896728515625
111 1.5009521484375
112 1.5114257812499998
113 1.5299560546875
114 1.5243164062499999
115 1.5347900390625
116 1.5355957031249998
117 1.525927734375
118 1.53076171875
119 1.5364013671874999
120 1.5347900390625
121 1.546875
122 1.5428466796875
123 1.55009765625
124 1.538818359375
125 1.329345703125
126 1.3954101562499999
127 1.4445556640625
128 1.482421875
129 1.5017578125
130 1.513037109375
131 1.5162597656249999
132 1.5218994140625
133 1.5347900390625
134 1.5436523437499998
135 1.5364013671874999
136 1.5355957031249998
137 1.533984375
138 1.5283447265624999
139 1.5396240234374998
140 1.542041015625
141 1.55009765625
142 1.5484863281249999
143 1.329345703125
144 1.1915771484375
145 1.3140380859374998
146 1.2536132812499998
147 1.3478759765625
148 1.4171630859375
149 1.3921875
150 1.3849365234375
151 1.3913818359374999
152 1.3986328124999998
153 1.426025390625
154 1.4469726562499998
155 1.4550292968749998
156 1.416357421875
157 1.4429443359374998
158 1.4252197265625
159 1.4228027343749998
160 1.4252197265625
161 1.44052734375
162 1.4542236328125
163 1.445361328125
164 1.4638916015624999
165 1.4558349609374999
166 1.44052734375
167 1.4526123046875
168 1.4381103515625
169 1.4268310546874998
170 1.4252197265625
171 1.4123291015625
172 1.4091064453125
173 1.4066894531249998
174 1.4268310546874998
175 1.4219970703125
176 1.4042724609375
177 1.405078125
178 1.426025390625
179 1.4413330078125
180 1.44052734375
181 1.4574462890625
182 1.4244140625
183 1.461474609375
184 1.461474609375
185 1.474365234375
186 1.47275390625
187 1.4276367187499999
188 1.4501953125
189 1.4461669921875
190 1.4284423828125
191 1.4171630859375
192 1.416357421875
193 1.4445556640625
194 1.42119140625
195 1.26005859375
196 1.332568359375
197 1.3688232421875
198 1.3913818359374999
199 1.384130859375
200 1.4139404296875
```

rwaldron, 2016-06-21:
I have also experienced this behavior

Frijol, 2017-02-13:
Due to no comments since https://github.com/tessel/t2-firmware/pull/204, I think the issue is resolved. Please reopen if you come across the same issue.

closed 2017-02-13 by Frijol