tessel/t2-firmware #169: Received unexpected response with no commands pending: 0

jendib, 2016-04-19:
After running [this code](https://github.com/sismics/home/blob/master/tessel/app.js) for some hours, I get this error:
```
/usr/lib/node/tessel-export.js:271
          throw new Error('Received unexpected response with no commands pending: ' + byte);
          ^

Error: Received unexpected response with no commands pending: 0
    at null.<anonymous> (/usr/lib/node/tessel-export.js:271:17)
    at emitNone (events.js:67:13)
    at Socket.emit (events.js:166:7)
    at emitReadable_ (_stream_readable.js:411:10)
    at emitReadable (_stream_readable.js:405:7)
    at readableAddChunk (_stream_readable.js:157:11)
    at Socket.Readable.push (_stream_readable.js:110:10)
    at Pipe.onread (net.js:523:20)
```

I'm guessing it's related to my usage of the serial port.

I'll be happy to fix this myself, but can you point me in the right direction please?
I don't really understand the error.

Thanks.

johnnyman727, 2016-04-20:
@jendib thanks for stress testing the firmware :)

This error is [thrown here](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L271). The SAMD21 processor relays data between the modules (or, more specifically, the hardware interfaces like UART) to the MediaTek processor via a SPI connection to an awaiting daemon. That daemon then relays it further to the specific domain socket that the port is using. The Tessel JavaScript driver then parses that data to determine the source and the content (ie "a GPIO was raised high on port A" or "Received these 5 bytes of data over UART on port B"). This error is saying that the JavaScript driver received a byte that was parsed according to the protocol to indicate that it was a command that should have had additional data bytes but those bytes were missing.

I think the first step towards debugging would be to use `t2 root` to get shell access to your Tessel, then modify `/usr/lib/node/tessel-export.js` to save the incoming data to some log file. Then we can see what the commands being received were and what they should have been. Or, more simply, you can just add an extra parameter to the console.log to print the value of `byte` (which indicates the command type) with the error. But I think the first suggestion is better because it gives us more context.

JakeHartnell, 2016-07-12:
+1, any progress on this?

I added a `console.log(byte);` before throwing the error in tessel-export.js and got:
> 132

Followed by:

```
/usr/lib/node/tessel-export.js:271
          throw new Error('Received unexpected response with no commands pending: ' + byte);
          ^

Error: Received unexpected response with no commands pending: 132
    at null.<anonymous> (/usr/lib/node/tessel-export.js:271:17)
    at emitNone (events.js:67:13)
    at Socket.emit (events.js:166:7)
    at emitReadable_ (_stream_readable.js:419:10)
    at emitReadable (_stream_readable.js:413:7)
    at readableAddChunk (_stream_readable.js:164:13)
    at Socket.Readable.push (_stream_readable.js:111:10)
    at Pipe.onread (net.js:531:20)

```

PR #204 (2016-09-02): Stability fixes
See individual commit messages for details.

My primary test was running the accelerometer demo on port A and ambient demo on port B with `t2 run --usb`, which fails within seconds on the current release, and hasn't failed yet with these changes.

Fixes #183, Fixes #169 (at least three of these bugs could lead to that kind of corrupted response)
Possibly #128
Possibly #166
closed 2016-09-02 by johnnyman727