tessel/t2-firmware #172: Add names to existing functions

wyze, 2016-04-27:
Examples
---

`Tessel.Port`: https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L166
```diff
-Tessel.Port = function(name, socketPath, board) {
+Tessel.Port = function TesselPort(name, socketPath, board) {
```

`Tessel.Pin`: https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L503
```diff
-Tessel.Pin = function(pin, port, interruptSupported, analogSupported, pullSupported, pwmSupported) {
+Tessel.Pin = function TesselPin(pin, port, interruptSupported, analogSupported, pullSupported, pwmSupported) {
```
---
Stacktraces
---
This will give better stack traces as seen below:

Before:
```
~/Projects/tessel/ambient-attx4 master
❯ t2 run examples/ambient.js
INFO Looking for your Tessel...
INFO Connected to Tessel-02A30510B2C5.
INFO Building project.
INFO Writing project to RAM on Tessel-02A30510B2C5 (42.496 kB)...
INFO Deployed.
INFO Running examples/ambient.js...
/usr/lib/node/tessel-export.js:180
  throw new Error('Simulated Error');
  ^

Error: Simulated Error
    at new Tessel.pwmFrequency.Tessel.Port (/usr/lib/node/tessel-export.js:180:9)
    at new Tessel (/usr/lib/node/tessel-export.js:59:26)
    at Object.<anonymous> (/usr/lib/node/tessel.js:3:18)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/remote-script/examples/ambient.js:1:69)
```

After:
```
~/Projects/tessel/ambient-attx4 master
❯ t2 run examples/ambient.js
INFO Looking for your Tessel...
INFO Connected to Tessel-02A30510B2C5.
INFO Building project.
INFO Writing project to RAM on Tessel-02A30510B2C5 (42.496 kB)...
INFO Deployed.
INFO Running examples/ambient.js...
/usr/lib/node/tessel-export.js:180
  throw new Error('Simulated Error');
  ^

Error: Simulated Error
    at new TesselPort (/usr/lib/node/tessel-export.js:180:9)
    at new Tessel (/usr/lib/node/tessel-export.js:59:26)
    at Object.<anonymous> (/usr/lib/node/tessel.js:3:18)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/remote-script/examples/ambient.js:1:69)
```
---
Safeguards
---
This will also help with safeguards like this: 

```js
function use(hardware, callback) {
  if (!hardware) {
    console.warn(new Error('Improperly specified Tessel port.'));
  }

  return new Ambient(hardware, callback);
}
```

From https://github.com/tessel/ambient-attx4/blob/master/index.js#L355

So instead of checking if something is passed in, it could be updated to something like:

```js
function use(hardware, callback) {
  if (
    typeof hardware !== 'object' ||
    (typeof hardware === 'object' && hardware.constructor.name !== 'TesselPort')
  ) {
    console.warn(new Error('Improperly specified Tessel port.'));
  }

  return new Ambient(hardware, callback);
}
```

rwaldron, 2016-04-27:
I have a lot of thoughts about this and most of them are in opposition. I'm mobile at the moment, but I'd like some time to reply before you commit to writing this patch. I don't think it's necessary, or wise, to write guards that key on some arbitrary constructor.name string value. In this example given, you've already broken t1-firmware compatibility, because that thing is called `Port`. Sure, you can change it, but then it's just whack-a-mole. 

ES2015 explicitly specified the value of all function object's name properties—if V8 isn't resolving those properly, then it's a bug in V8, not this project.

rwaldron, 2016-04-27:
Also, using constructor.name precludes writing singleton port objects (eg stubs, mocks, etc).

wyze, 2016-04-28:
> I'm mobile at the moment, but I'd like some time to reply before you commit to writing this patch.

Of course, that's why the issue is open instead of a PR, to get some discussion to not make wasted work. :)

> I don't think it's necessary, or wise, to write guards that key on some arbitrary constructor.name string value.

The main driver behind this would be to help facilitate tessel/ambient-attx4#63. So if you could chime in on that issue with some of your thoughts, that would be awesome!

> In this example given, you've already broken t1-firmware compatibility, because that thing is called Port. Sure, you can change it, but then it's just whack-a-mole.

> ES2015 explicitly specified the value of all function object's name properties—if V8 isn't resolving those properly, then it's a bug in V8, not this project.

I think V8 is assigning a name fine if you look at the before stacktrace: 

```
at new Tessel.pwmFrequency.Tessel.Port (/usr/lib/node/tessel-export.js:180:9)
```

> Also, using constructor.name precludes writing singleton port objects (eg stubs, mocks, etc).

The stubs/mocks did not cross my mind. I need to remember to put this into my thought process when I am coming up with these ideas.

wyze, 2016-04-29:
Closing this issue in favor of a better solution discussed here: https://github.com/tessel/ambient-attx4/issues/63#issuecomment-215444612

closed 2016-04-29 by wyze