tessel/t2-firmware #173: I2C.send calls callback with extra argument

johnnyman727, 2016-05-17:
I expect the callback of `I2C.send` to just have an `error` attribute but it also has a second argument with the value of `new Buffer([0x88])`.

Dr-Emann, 2016-10-02:
Looks like [`I2C.send`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L924) calls the callback with `_simple_cmd(CMD.STOP)`, whereas [`I2C.read`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L931)/[`I2C.transfer`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L943) call the callback with `port.rx`, and before sending STOP.

rwaldron, 2016-10-02:
Send only sends and then stops. Read and transfer must wait for responses before the stop.

Dr-Emann, 2016-10-02:
Neither does [`pin.high()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L698), but it accepts a callback parameter.

Dr-Emann, 2016-10-02:
Functions which also pass `[0x88]` to their callback parameter:

* [`pin.high()`]
* [`pin.low()`]
* [`pin.toggle()`]
* [`pin.input()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L762)
* [`pin.pull()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L794)
* [`spi.send()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L1008)
* [`uart._write()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L1079) (which doesn't appear to be used anywhere?)

[`pin.high()`]: https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L699
[`pin.low()`]: https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L704
[`pin.toggle()`]: https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L719