tessel/t2-firmware #174: pin.analogWrite inconsistency: no callback parameter

johnnyman727, 2016-05-17:
https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L752

Dr-Emann, 2016-10-02:
The link is now out of date. Press y when viewing code on github to go to a canonical link which won't change over time. Here's the [analogWrite function](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L824)

rwaldron, 2016-10-02:
@johnnyman727 is there a response from an analog write? I don't think there is

Dr-Emann, 2016-10-02:
There's no response from [`pin.high()`](https://github.com/tessel/t2-firmware/blob/fe9165c7137a8c0fe5809c282e39f8f267ec92e1/node/tessel-export.js#L698) either, but it has a callback parameter.