tessel/t2-firmware #175: analog pin values

Frijol, 2016-06-05:
It's currently confusing how analog pin values work. I think they are settable/readable in the range of 0 - 3.3 to match the voltage output/input.

User expectation (and [our current documentation](https://github.com/tessel/t2-docs/blob/master/hardware-api.md#analog-pins)) is that it should be settable in the range of 0-1 as a percentage of 3.3V.

Discussing with @flaki and @johnnyman727, @flaki suggests we have the current API use the percentage implementation and add a second API call like "analogWriteRaw" which has you set as 0-3.3.

@rwaldron @tcr you may have opinions on this?

rwaldron, 2016-06-06:
I have a lot of opinions, but most of them require a time machine to resolve... The actual hardware function is a 10-bit DAC write, so we should just make a `dacWrite` function that accepts a 10-bit value. I thought this analogWrite was sufficiently bogus when I was writing Tessel-IO for Johnny-Five, [so I just stepped around it](https://github.com/rwaldron/tessel-io/blob/master/lib/tessel.js#L352). My assumption was that it was somehow necessary for backwards compatibility, but I never bothered to look until just now and it appears that is not the case.

Frijol, 2016-06-16:
Good point. We should probably make it match whatever T1 did. Which I think is read a percentage. Thought [the T1 docs](https://github.com/tessel/t1-docs/blob/master/hardware-api.md#pins) would clarify, but it's not noted.

@rwaldron do you have a T1 handy that you could read something 3.3V with and see if it gives you a 1 back?

rwaldron, 2016-07-05:
@Frijol sorry I don't have a functioning T1 anymore :\

dudleyjosh, 2016-07-05:
Analog returned 0 to 1 on T1... from the T1 docs:

> Analog pins can be any value between 0 and 1 (GND/0V and 3.3V).

rwaldron, 2016-07-06:
Ok, so the present state is kind of a complicated mess so I'm going to write it all out here...

1. `analogRead`
  1. The T1 returns a value, 0-1, that represents an "analog voltage measurement" as a percentage of 3.3V
    1. This matches the documentation
  2. The T2 returns a value, 0-3.3, that represents an "analog voltage measurement" as an actual level within 3.3V.
    1. The documentation says this should be 0-1 (an "analog voltage measurement" as a percentage of 3.3V)
  3. Both of these diverge from the expected value produced by an "analog read" operation—which is generally a value within some _range_, where _range_ is 0-_n_ and _n_ is the maximum resolution of the hardware, eg. 10-bit = 1023, 12-bit = 4096, etc.
2. `analogWrite`
  1. This accepts a value, 0-3.3:
    ```js
    // From the source...
    var data = val / 3.3 * 0x3ff;
    ```

    1. This contradicts the docs, which describe a value, 0-1, that represents an "voltage" as a percentage of 3.3V: 

      ```js
      // From the docs...
      pin.analogWrite(0.6);  // turn pin to 60% of high
      ```

  1. This diverges from the expected value to be written by an "analog write" operation—which is generally a value that can be described in the same terms as 1.iii.

It's unclear why T2's `analogRead` was defined in contradiction to the T1, hopefully @kevinmehall can offer more insight. 

The fixes necessary are as follows: 

1. `analogRead`
  1. This line: https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L745 needs to have the ` * 3.3` removed. The tests must also be updated. 
2. `analogWrite`
  1. To by symmetrical with `analogRead`, this line: https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L759 needs to have ` / 3.3` removed. The tests must also be update.

rwaldron, 2016-07-06:
If this doesn't have a taker by Monday, we should assign someone.

Frijol, 2016-07-07:
quick note: @johnnyman727 researched further & it was @jiahuang who implemented (not @kevinmehall) so we are seeking clarification from her in case there's a good reason it was done this way

rwaldron, 2016-07-07:
Thanks for the additional clarification/follow-up :)

rwaldron, 2016-07-07:
Nominating @carlbaron to take this.

carlbaron, 2016-07-08:
@rwaldron Sounds great! I'm excited to dive into this!

johnnyman727, 2016-07-14:
Confirmed with the original author that there was no special reason for the 0-3.3 scale so we're golden.

Frijol, 2016-09-02:
resolved with https://github.com/tessel/t2-firmware/pull/184

closed 2016-09-02 by Frijol