tessel/t2-firmware #188: Reboot function call

reconbot, 2016-07-30:
I'd like to propose a `tessel.reboot()` function call on the tessel object. You can shell out to reboot, but that seems messy.

rwaldron, 2016-08-04:
Reboot isn't related to port & io interaction—why not just let user code shell out to `reboot`?

reconbot, 2016-08-04:
Do we want people using linux or js? We provide a bunch of tessel services
this seems reasonable to provide.

rwaldron, 2016-08-04:
Is rebooting directly from your JS code something that is frequently needed?

reconbot, 2016-08-04:
It's something I do to my router and my nest and at least once to my drop
cam when something hasn't worked right.

HipsterBrown, 2016-08-04:
@reconbot How do you see if being used in your programs? Is it something called by a physical button (but not the on-board one) or from some kind of web UI? 

`reboot` seems like a nuclear option since it tears down the entire program and that should be very clearly communicated to the Tessel developers.

reconbot, 2016-08-04:
If I'm a developer prototyping an app I don't see a major use case. 

If I'm a user of a tessel based app and the tessel is plugged in somewhere and hidden away and it starts to act funky then I see the value in this functionality. And if I'm a developer thinking of my user, then I see the value in the function call.

I kind of think we should be able to provide programmatically all the functionality of the tessel. We've hidden the openwrt backend from normal use and I think we should keep it hidden and continue to provide a nice clean api.

dtex, 2016-08-04:
Some percentage of users have limited exposure to *nix (i.e. Windows and some Mac users) and as soon as you tell them to "just shell out to ```reboot```" you've disrupted the amazing, simple developer experience that the Tessel 2 offers.

rwaldron, 2016-08-04:
I'll spend some time investigating this

closed 2016-09-14 by rwaldron