tessel/t2-firmware #191: Wifi isn't initialized with connection information on startup

reconbot, 2016-07-31:
When I use `tessel.network.wifi.isConnected` or `tessel.network.wifi.connection()` via `t2 run --lan` on a wifi connected tessel I get no connection information. If we reconnect via the api `isConnected` and `connection()` return the expected values.

Steps to reproduce:

```
⚡ tessel-wifi-setup$ t2 wifi
INFO Looking for your Tessel...
INFO Connected to eva.
INFO Connected to "TheHouse"
INFO IP Address: 192.168.0.128
INFO Signal Strength: (51/70)
INFO Bitrate: 72mbps
⚡ tessel-wifi-setup$ t2 run --lan index.js 
INFO Looking for your Tessel...
INFO Connected to eva.
INFO Building project.
INFO Writing project to RAM on eva (3.584 kB)...
INFO Deployed.
INFO Running index.js...
isConnected false
connection null
⚡ tessel-wifi-setup$ cat index.js 
var tessel = require('tessel');
console.log('isConnected', tessel.network.wifi.isConnected);
var currentConnection = tessel.network.wifi.connection();
console.log('connection', currentConnection);
```

HipsterBrown, 2016-07-31:
Thanks for making this issue @reconbot. It's a duplicate of this existing issue -> #178. Closing but adding a reference to this issue in the comments of the first one.

closed 2016-07-31 by HipsterBrown

reconbot, 2016-07-31:
Oh sweet thanks!