tessel/t2-firmware #194: Support for disabling power to a module

njbuch, 2016-08-02:
For future release we should consider a feature that supports (or encourages) module builders to control the power to the module. To ensure really low power scenarios it would be nice to fully control the power consumption of the setup in code.

There is certainly challenges in regard to building a proper driver, as the on/off activity could be tricky to support for certain ICs and circuits.

Frijol, 2016-08-02:
cc @ekolker I think you had some ideas about this some time ago; @rwaldron may also be interested in this topic

kevinmehall, 2016-08-21:
Power to the module port is disabled when the port socket is closed. See https://github.com/tessel/t2-firmware/pull/201.

closed 2016-09-02 by Frijol