tessel/t2-firmware #199: [Proposal] Sleep ability

reconbot, 2016-08-09:
Both chips support going to sleep, so does open-wrt. The hardware also seems to support being woken up by a timer or an input pin. Once we have a some more developed network events this could be very valuable for applications that don't need to always be on.

Hardware aside, I don't know how nodejs and open wrt support sleep. Can we have a call that pauses node until it's woken up? Or will node keep running until the processes is itself suspended.

I'm not suggesting this be the actual API just demonstrating possible uses.

```js
function readLoop(){ 
  SensorNetwork.takeReadings()
    .then(savetoCloud)
    .then(() => tessel.sleep(30 * 60 * 1000))
    .then(readLoop);
}
```

```js
tessel.wakeOnPin(4);
tessel.sleep();
console.log('pin 4 has woken me!');
```

rwaldron, 2016-08-09:
> Hardware aside, I don't know how nodejs and open wrt support sleep. 

Irrelevant to openwrt, Node.js can't do "sleep"

> Can we have a call that pauses node until it's woken up? 

Nope

> Or will node keep running until the processes is itself suspended.

That's correct. Or...

- When there are no more tasks left in the current execution turn and no future execution turns scheduled.
- Process exit by signal
- Power loss

kevinmehall, 2016-08-21:
Duplicate of #14.

MT7620 does not have a "Sleep" mode that preserves RAM contents like sleep mode on a laptop. The capability we do have is that the SAMD21 controls the power to the MT7620, so we can shut down Linux, cut power to the MT7620, ports, and USB, and then put the SAMD21 to sleep pending an interrupt for a timer or pin change. On that event, it would restore power to the MT7620, which would go through the normal boot process. The Node app would start from the beginning, and would have to save any state that it wants to preserve to flash.

closed 2016-08-21 by reconbot

reconbot, 2016-08-21:
Thanks @kevinmehall I'll close this in favor of #14. That would be interesting to program for...