tessel/t2-firmware #205: Led returns empty object

Connor-Knabe, 2016-08-26:
I attempted to turn off the leds per the docs instructions.

```
var tessel = require('tessel'); // Import tessel

var leds = tessel.led;

console.log('leds', leds);

```

Console prints:

leds {}

johnnyman727, 2016-08-26:
What does `leds[0]` get you? Does the script created by `t2 init` run properly?

Connor-Knabe, 2016-08-26:
It also gets me an empty object.  I tried t2 init and it seems to blink fine but I cannot get it to turn the leds off. Very strange.

johnnyman727, 2016-08-26:
I'm guessing this is an issue with how the object is being printed, not the actual structure of the leds. `leds[2].low()` should turn them off.

rwaldron, 2016-08-27:
I don't recommend using `led.low()`, because it expects a callback (legacy?), just use `on()` and `off()`. The `tessel.led` object would fail its unit tests if it was an empty object. 

https://github.com/tessel/t2-firmware/blob/master/node/test/unit/tessel.js#L260

closed 2016-08-27 by rwaldron

Connor-Knabe, 2016-08-29:
@rwaldron any idea on why the LEDs would not turn off? I called `off()` but it didn't seem to actually turn them off.