tessel/t2-firmware #206: Upgrade to Node 6

Frijol, 2016-09-01:
Per 
[image]

It's going to LTS in October, and we should move with it

tcr, 2016-09-08:
Some tasks:

* [ ] Switch OpenWRT build to use LTS branch http://github.com/tessel/openwrt-tessel/pulls/58 
* [ ] Disable iconv (cc @rwaldron)
* [ ] Create precompiled modules for each version, have t2-cli pull them down appropriately

tcr, 2016-09-08:
@huseyinkozan has these two branches to support porting:

* https://github.com/huseyinkozan/t2-firmware/commits/master
* https://github.com/huseyinkozan/openwrt-tessel/commits/node-6.2.1

rwaldron, 2016-09-08:
It's more than just iconv, I believe we need to disable intl to avoid inclusion of icu data. https://github.com/nodejs/node/blob/master/configure#L322-L326 

@srl295 can you advise?

tcr, 2016-09-08:
Also @huseyinkozan you did this in your commit https://github.com/huseyinkozan/openwrt-tessel/commit/4d20a94901ce1e61447d45b120ef92dc8462c034, was that sufficient to bring the size of the Node binary down?

rwaldron, 2016-09-08:
> Also @huseyinkozan you did this in your commit huseyinkozan/openwrt-tessel@4d20a94, was that sufficient to bring the size of the Node binary down?

That looks correct to me, but I would like @srl295 to confirm as well

rwaldron, 2016-09-08:
@srl295 directed me to: https://github.com/nodejs/node/blob/master/BUILDING.md#building-without-intl-support which confirms that @huseyinkozan's patch is correct

huseyinkozan, 2016-09-08:
Hi all,
@tcr, it is 19 MB. Dont remember the size gain. It just works. The binary below built by the commit you noted. I tried to include libicu and enable it again, but no success.

http://huseyinkozan.com.tr/files/tmp/openwrt-ramips-mt7620-tessel-squashfs-sysupgrade-node-6.2.1-without-intl.bin

Our app is so large, and it works on my build. These are the deps:
```
  "dependencies": {
    "body-parser": "^1.15.1",
    "buffer-layout": "^0.13.0",
    "compression": "^1.6.2",
    "cookie-parser": "^1.4.3",
    "diskspace": "^1.0.3",
    "express": "^4.14.0",
    "express-session": "^1.13.0",
    "fs-extra": "^0.30.0",
    "mathjs": "^3.3.0",
    "moment-timezone": "^0.5.4",
    "multer": "^1.1.0",
    "node-json-db": "^0.7.2",
    "passport": "^0.3.2",
    "passport-http": "^0.3.0",
    "pidusage": "^1.0.1",
    "semantic-ui": "^2.1.8",
    "serialport": "^3.1.2",
    "session-file-store": "^0.2.0",
    "string": "^3.3.1",
    "touch": "^1.0.0"
  }
```
Serialport library is for PC port. The app serve a semantic-ui GUI, an API on expressjs, and an apidoc static directory. 

I want to show the demo site, but my bosses want to wait unil finalize our product.  : ^ /

I will try to update from upstream and build it again to night (UTC+3).

huseyinkozan, 2016-09-08:
Previous works:
https://github.com/tessel/openwrt-tessel/pull/58

johnnyman727, 2016-09-08:
Just a note that the Flash size went down to ~14.5MB when [we swapped ffmpeg for mpeg-streamer](https://github.com/tessel/openwrt-tessel/pull/65/files) in `0.0.14`.

rwaldron, 2016-09-08:
@huseyinkozan rad :)

huseyinkozan, 2016-09-08:
@johnnyman727, how much before ? Is this for app usage, not for all OpenWRT ?

johnnyman727, 2016-09-08:
@huseyinkozan it was at about 19MB before. This is just the base OpenWRT image that sits in Flash. App usage will vary by app.

srl295, 2016-09-09:
hi all! @huseyinkozan you said
>  I tried to include libicu and enable it again, but no success

can you get me more details? Sounds like a separate issue is in order at this point, `--without-intl` is probably the best step at this point.

huseyinkozan, 2016-09-09:
Sorry @srl295, didnt see your message at here. I wrote a detailed one at https://github.com/nodejs/node/issues/7173#issuecomment-245799611. But that was not the case that you quote. 

I had build libicu with OpenWRT environment, but couldnt say node to use this at OpenWRT packaging file.

libicu package Makefile for OpenWRT:
https://github.com/huseyinkozan/openwrt-tessel/blob/node-6.2.1/package/openwrt-addpack/icu/Makefile

Node package Makefile for OpenWRT:
https://github.com/huseyinkozan/openwrt-tessel/blob/node-6.2.1/package/node/node/Makefile

huseyinkozan, 2016-09-10:
Yes. I give up with icu. I built with --without-intl option. My trials are here :
https://github.com/huseyinkozan/openwrt-tessel/tree/node-6.5.0

Built files :
http://huseyinkozan.com.tr/files/tmp/

process.versions:
```
{ http_parser: '2.7.0',
  node: '6.5.0',
  v8: '5.1.281.81',
  uv: '1.9.1',
  zlib: '1.2.8',
  ares: '1.10.1-DEV',
  modules: '48',
  openssl: '1.0.2d' }
```

huseyinkozan, 2017-05-24:
https://github.com/tessel/openwrt-tessel/pull/73

huseyinkozan, 2017-07-24:
Still open ?

rwaldron, 2017-07-25:
Whoops! Thanks for follow up

closed 2017-07-25 by rwaldron