tessel/t2-firmware #208: Why do both module ports turn on when either module port is used?

Frijol, 2016-09-03:


kevinmehall, 2016-09-03:
tessel.js opens both ports when you merely `require('tessel')`

johnnyman727, 2016-09-03:
You can specify if you only want one or the other with, for example:
```.js
require('tessel-export')({
  ports: { B: false}
});
```

Frijol, 2016-09-03:
Interesting. This seems counterintuitive – why not disable by default and then enable when called upon?

johnnyman727, 2016-09-03:
I think mostly for backwards compatibility - the T1 powered both ports by default.

kevinmehall, 2016-09-03:
I don't think it was intentional, just that the constructor was the most convenient place to open the socket, and the constructor gets called on `require()` because the port is a property of the module (vs explicitly creating / opening the port with a function call or constructor)

rwaldron, 2016-09-14:
> I think mostly for backwards compatibility - the T1 powered both ports by default.

Confirm—and now that T2 has shipped, it remains for compatibility with itself, as literally all existing code would break if this changed. 

-----

Since the original issue title isn't correct, and there are now a couple ways to control port power manually, I think this can be closed. 

Control port power manually: 

```js
var tessel = require('tessel');

tessel.close("A"); // A is now closed
```

If a program requires port power control from the start of the program and the power must be initially "off" (which is likely not the most common case), then it's a little extra effort, but completely possible: 

```js
var Tessel = require('tessel-export');

var tessel = new Tessel({
  ports: { 
    A: false,
    B: false, 
  }
});

tessel.open("A"); // A is now open
tessel.open("B"); // B is now open
```

Maybe instead of closing this, we can update it to be about documentation?

Frijol, 2016-09-14:
Good call, and I'll move it over to t2-docs

Frijol, 2016-09-14:
This issue was moved to tessel/t2-docs#117

closed 2016-09-14 by Frijol