tessel/t2-firmware #212: chipSelectActive in SPI isn't actually used when toggling CS

tcr, 2016-09-05:


johnnyman727, 2016-09-06:
See the `chipSelectActive` property [in the Tessel 1 docs](https://github.com/tessel/t1-docs/blob/master/hardware-api.md#api-new-port-SPI-options) for a description of how that property should work. It should be added [to the SPI implementation of Tessel 2](https://github.com/tessel/t2-firmware/blob/31c441decbd9f20b918e2c67fb52b962167dee88/node/tessel-export.js#L911-L1000).

brihijoshi, 2017-07-27:
Hi, we were having a look at this issue. Isn't this issue already implemented in [this](https://github.com/tessel/t2-firmware/blob/31c441decbd9f20b918e2c67fb52b962167dee88/node/tessel-export.js#L918-L926) part of the code?

rwaldron, 2017-07-27:
@brihijoshi nice follow up work! We can go ahead and close this now :)

closed 2017-07-27 by rwaldron