tessel/t2-firmware #213: No way to disable chipSelect in SPI constructor

tcr, 2016-09-05:
I am of the belief that the SPI constructor should default to having no chip select by default, but at the least I should be able to set the constructor to null and have it not select one.

johnnyman727, 2016-09-06:
> I am of the belief that the SPI constructor should default to having no chip select by default

+ 1. I think we made a bad call on this for T1. Not sure how many code bases it would break (we'd definitely need to update a few module libraries), but I'd be interested in seeing it switched to no CS unless specified.

johnnyman727, 2016-09-06:
> I am of the belief that the SPI constructor should default to having no chip select by default

+1 - I think we made a bad call on this for T1. Not sure how many code bases it would break (we'd definitely need to update a few module libraries), but I'd be interested in seeing it switched to no CS unless specified.