tessel/t2-firmware #223: Support UART baud rate up to 230400

jeffyactive, 2017-01-18:
Propose to change **node/tessel-export.js** as follows (~lines 1055-1057):

```
        if (value < 9600 || value > 230400) {
          throw new Error('UART baudrate must be between 9600 and 230400');
        }
```

We have successfully validated that 230400 bps operation is supported on the Tessel 2.  The previous upper bound was 115200 bps.  Perhaps even higher baud rates are possible, but we haven't tested.

rwaldron, 2017-01-18:
I can't find any source material in the datasheet to support the 115200 limit. @kevinmehall can likely provide us the insight we seek.

kevinmehall, 2017-01-18:
SAMD21 hardware should support UART baud rates of several Mbaud, but I haven't checked if the firmware would need to change divisor settings on the SERCOM clock to be able to use higher speeds.

jeffyactive, 2017-01-18:
We're using 230400bps in production right now by manually editing tessel-export.js as specified.  Agree that ideally the upper baud rate would be pushed to whatever the SAMD21 supports without changing divisor settings.

jeffyactive, 2018-02-15:
Friendly bump as I've now manually changed the baud rate limit on over 250 Tessels and would really appreciate this simple change applied in the next firmware release.  @rwaldron and @kevinmehall 
 please advise if I should:
- make a pull request of the code in my original post
- try to figure out the actual max baud rate and then include that in the pull request
- do nothing and wait...

Many thanks!

HipsterBrown, 2018-10-22:
This has been fixed by @jeffyactive's PR: #242  🎉