tessel/t2-firmware #224: Programmatically switch from Access point to normal wifi

julesmoretti, 2017-02-02:
Is there a way to programmatically switch a tessel from access point to not in order to have the tessel then connect to a desired wifi network.

This is order to create an interaction where an iPhone app would connect to the tessel at first, set it to a wifi network and then let it connect to it. 

All this only using your phone...

HipsterBrown, 2017-02-15:
Hey @julesmoretti,

Sorry about just a delayed response. You can use the `tessel` [Network API](https://tessel.gitbooks.io/t2-docs/content/API/Network_API.html) to start an access point when it boots, if it's not connected to a wifi network. You can serve a web page from Tessel that allows a person to choose their desired wifi network, then use that info with the Network API to connect the Tessel to it. Let me know if you have any questions about using the API.

huseyinkozan, 2017-07-24:
github.com/huseyinkozan/tessel-network-example