tessel/t2-firmware #235: Get i2c address

rwaldron, 2017-06-26:
Originally filed here: https://github.com/tessel/t2-cli/issues/1257

@opkiler22789: 
> Hello, Is there any way to get i2c address? In the likeness of [i2c_scanner](http://playground.arduino.cc/Main/I2cScanner)?

rwaldron, 2017-06-26:
There is no way to scan i2c addresses in the manner that `i2c_scanner` does, but I'm wondering if provide a similar (possibly simpler) alternative? I'm going to put together a rough sketch this week.

HipsterBrown, 2017-06-26:
Is this something that would mostly happen on the SAMD21 (the C firmware) with `event` messages back to the SoC (Node.js firmware)?

rwaldron, 2017-06-26:
Yes, that's correct. The JS API would just be a wrapper around some command that is sent to SAMD21, the command would do whatever test is necessary on the SAMD21 and respond with the outcome.

fivdi, 2017-06-26:
Would it not be possible to loop over all I2C addresses calling [i2cConfig](https://github.com/rwaldron/tessel-io/blob/master/lib/index.js#L774) and then [i2cReadOnce](https://github.com/rwaldron/tessel-io/blob/master/lib/index.js#L900) for each address. After calling `i2cReadOnce`, wait for either an [error](https://github.com/rwaldron/tessel-io/blob/master/lib/index.js#L929) to be emitted indicating that there is no device at the address or an [event](https://github.com/rwaldron/tessel-io/blob/master/lib/index.js#L941) to be emitted indicating that a there is a device at the address.

This is probably a bit of hack and it may leave [state.i2c](https://github.com/rwaldron/tessel-io/blob/master/lib/index.js#L817) full of junk that's not needed. Reading information from a device may also have unwanted side effects on the device. On the other hand it might work!

It's probably of little help, but i2c-bus has [scan](https://github.com/fivdi/i2c-bus/blob/v1.2.2/i2c-bus.js#L351-L379) and [scanSync](https://github.com/fivdi/i2c-bus/blob/v1.2.2/i2c-bus.js#L381-L396) methods that use the SMBus Receive Byte protocol to scan for devices. The Linux `i2cdetect -y -r <bus-number>` command also uses the SMBus Receive Byte protocol to scan for devices.

rwaldron, 2017-06-27:
@fivdi the JS runs on the mediatek mt7620n and communicates with the samd21 via domain socket. The entire i2c stack exists on samd21 and there is presently no response produced when a program attempts to write (via a message sent over the socket) to a non-existent i2c peripheral address. 

```
const tessel = require('tessel');

const txb = new Buffer([0xff]);
const i2c = new tessel.port.A.I2C(0x01);

i2c.transfer(txb, 1, (error, data) => {
  // Never executed, because no response ever occurs. 
});
```

rwaldron, 2017-06-27:
https://github.com/tessel/t2-firmware/blob/master/firmware/port.c#L454-L460