tessel/t2-firmware #238: readPulse not implemented

nheine, 2017-11-29:
Is there a plan to implement readPulse function in tessel-export.js in the near future? That would be great
Cheers

HipsterBrown, 2017-11-30:
@nheine What is the functionality of `readPulse` that you would like to see in the API?

nheine, 2017-11-30:
I expect the same behaviour than on tessel1 to read a PWM signal. Or is there another way for tessel2?

HipsterBrown, 2017-11-30:
Ah ok. We do not have anyone working on that feature at the moment. We are open to pull requests. I'll bring it up at the next Steering Committee meeting as well.

nheine, 2017-11-30:
would be great to get this fast. I am not sure how complicated it is to simply migrate the tessel1 code that already exists for this

HipsterBrown, 2018-12-18:
I still have not looked into how this would be implemented, and I don't have the experience to start working on something like this. If someone with C hardware experience can help port the functionality from the t1-firmware to work on the SAMD21, that's what we need.